MyoHub/myosuite #7: Upgrade to Mujoco 2.2.0

ViktorM, 2022-05-26:
Thanks for the release, great project! Do you consider upgrading your repo to the latest, open-sourced Mujoco 2.2.0 version?

Vittorio-Caggiano, 2022-05-26:
Hi @ViktorM, thanks for your support! 

There are a few aspects related to upgrading to MuJoCo 2.2

1. Update of the core MuJoCo from 2.1 (we are currently use) to 2.2 -- we use `free-mujoco-py` to simplify the installation process of `mujoco-py` and currently there is no support for MuJoCo 2.2

2. Move from `mujoco-py` to native mjc binding -- in the plans but not in the short terms

3. Provide binaries for the environment for MuJoCo 2.2 -- we are planning to release the full xml files which should be compatible with MuJoCo 2.x releases

Vittorio-Caggiano, 2023-07-13:
we are now supporting mujoco native binding. Please, let us know if they work well on your side

closed 2023-07-13 by Vittorio-Caggiano