MyoHub/myosuite #8: Release of MyoSim

sherwin0402, 2022-05-27:
Thank you for the great platform you built. May I ask if you are planning to open-source MyoSim as written in your paper? Thank you

HuaweiWang, 2022-05-27:
Thanks for your interest. Yes, open-source the MyoSim models, as well as the automatic model converting pipeline, is our next focus in the following half-year.

closed 2022-05-29 by Vittorio-Caggiano

vikashplus, 2022-05-31:
@sherwin0402: Currently available models can be found here -- https://github.com/facebookresearch/myosuite/tree/main/myosuite/envs/myo/assets

sherwin0402, 2022-05-31:
@vikashplus @HuaweiWang Thanks for the information. Are lower limb models like gait 2392, 2354 etc part of your plans for MyoSim as well? Or is upper limb the focus now?

HuaweiWang, 2023-08-10:
@sherwin0402 We officially released the model [conversion pipeline](https://github.com/MyoHub/myoconverter) today and it contains a few leg models converted as examples. Please have a look!