MyoHub/myosuite #14: Myosuit with windows

vahid34, 2022-06-03:
It seems that Myosuit only works on Linux and Mac. Does it also works on windows or not?

Vittorio-Caggiano, 2022-06-03:
Hi! we didn't test it on Windows (so we could not provide instructions). The major issue on running MyoSuite on windows is that we depend on `mujoco-py`, which is deprecated in windows. I know that some have successfully managed to install MyoSuite on windows, so it should be possible. 

The seps I would follow are:
1. install mujoco windows binaries -- https://github.com/deepmind/mujoco/releases/tag/2.1.2 under assets
2. pip install mujoco-py
3. git clone git@github.com:facebookresearch/myosuite.git
4. cd myosuite
5. pip install -e .

if you succeed, please suggest the steps you followed to update the [advanced installation instructions](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md)

Vittorio-Caggiano, 2022-07-18:
@vahid34 did the above solutions work for you?

GuillaumeVDu, 2022-07-21:
Hi,
maybe have a look here: [https://medium.com/@sayanmndl21/install-openai-gym-with-box2d-and-mujoco-in-windows-10-e25ee9b5c1d5](https://medium.com/@sayanmndl21/install-openai-gym-with-box2d-and-mujoco-in-windows-10-e25ee9b5c1d5)

Pouyan97, 2023-04-19:
Myosuite is only installed on Macbook or Linux. Therefore, installing Myosuite requires Linux on Windows. Two options available are the built-in WSL or a virtual box. The virtual box performs better for Myosuite due to its simpler visualization capabilities. 

1. Install Virtual box from [Here](https://www.virtualbox.org/wiki/Downloads)
2. Follow the instructions to create a virtual box with Linux Ubuntu 20.0.04+ installed (need to download the ubuntu .iso file and include it in the creation of the VM)
3. Follow the  linux [advanced installation](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md) guide to install Myosuite 
4. Download the github repo for myosuite to be able to run the python test code on your VM

**Potential issues:** 
- `IMPORT ERROR: ‘GLIBCXX_3.4.30 not found’`
Solution: solution found on stackoverflow: libstdcxx is replaced by libstdc++ in conda therefore we can need to libstdcxx-ng=12 for conda
Use the command:
`conda install -c conda-forge libstdcxx-ng=12`
If that doesn’t solve the issue refer to solutions presented [HERE](https://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error)

- `Illegal instructions (CORE DUMPED)` when running the test script provided in github page:
Solution: this issue might not be related to virtual box and windows itself. If you see a turtle with the V logo in the bottom right corner of your VM, these solutions and instructions can be found [HERE](https://stackoverflow.com/questions/65780506/how-to-enable-avx-avx2-in-virtualbox-6-1-16-with-ubuntu-20-04-64bit)

closed 2023-05-20 by Vittorio-Caggiano

Balint-H, 2023-05-28:
> Hi! we didn't test it on Windows (so we could not provide instructions). The major issue on running MyoSuite on windows is that we depend on `mujoco-py`, which is deprecated in windows. I know that some have successfully managed to install MyoSuite on windows, so it should be possible.
> 
> The seps I would follow are:
> 
> 1. install mujoco windows binaries -- https://github.com/deepmind/mujoco/releases/tag/2.1.2 under assets
> 2. pip install mujoco-py
> 3. git clone [git@github.com](mailto:git@github.com):facebookresearch/myosuite.git
> 4. cd myosuite
> 5. pip install -e .
> 
> if you succeed, please suggest the steps you followed to update the [advanced installation instructions](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md)

These steps worked for me for running on Windows without a VM, as far as getting to the random action example scenes, and passing tests. The following additional changes were required:

- `os.add_dll_directory(r"<PATH_TO_MINICONDA>\Miniconda3\envs\myosuite\lib\site-packages\mujoco_py\binaries\windows\mujoco210\bin")` added to `mujoco-py`'s `builder.py`
- removing `from os import sendfile`  (which is seemingly unused) from the `pen_v0.py` environment