MyoHub/myosuite #25: Issue running example.

nicos1993, 2022-09-18:
Hello, 

I am having issues with running the examples. My Ubuntu version is 22.04.1.
I am doing the following:
conda create -n "my_new_env" python=3.7.1 ipython
conda activate my_new_env
pip install free-mujoco-py (I was previously trying to install the latest version which likely was an issue!)
pip install -U myosuite

I then run into 2 errors which are somewhat identical:
/home/nicos/miniconda3/envs/my_new_env/lib/python3.7/site-packages/mujoco_py/gl/osmesashim.c:1:10: 
fatal error: GL/osmesa.h: no such file or directory; compilation terminated

Any advice to overcome these issues are greatly appreciated!

Thanks, 

Nicos

Vittorio-Caggiano, 2022-09-18:
Hi @nicos1993 

this seems to be a problem with `mujoco_py`. Did you try the instructions in the [advanced installation](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md#linux)?

Best,
Vittorio

nicos1993, 2022-09-19:
Hello @Vittorio-Caggiano 

Thanks for getting back to me. I hadn't come across the page you shared for installation. I was following the guidance provided here: https://myosuite.readthedocs.io/en/latest/install.html 
I don't think I would've come across the page you shared without pointing me to it! I will give those instructions an attempt.
Can you confirm whether I need to install the free-mujoco-py package? or mujoco-py? 

Thanks, 
Nicos

Vittorio-Caggiano, 2022-09-19:
I suggest `free-mujoco-py` because it should be easier to install. In principle you should get everything with the following commands:
``` bash
conda create --name MyoSuite python=3.7.1
conda activate MyoSuite
pip install -U myosuite
```

In the way `myosuite` is packaged, `free-mujoco-py` should be installed automatically.

thanks for the pointer to the missing information. I updated the rtd documentation with the link to the advanced installation

nicos1993, 2022-09-19:
I am following the instructions from the advanced installation; Perhaps I am being stupid but after I update conda I am asked to navigate to the myosuite folder prior to setting up the environment with the .yml file - Should I already have pip installed -U myosuite in my conda base environment? 

I will try running the 3 lines above too. 

Thanks, 
Nicos

Vittorio-Caggiano, 2022-09-19:
Try first the 3 lines of code I sent you. Those should be enough in most of the cases.

Regarding the advanced installation, you will need [conda](https://docs.conda.io/en/latest/miniconda.html#latest-miniconda-installer-links) and then, in the [env.yml](https://github.com/facebookresearch/myosuite/blob/main/setup/env.yml) file there is a check that you have pip installed.

nicos1993, 2022-09-19:
I have to first install `myosuite` prior to being able to run the env.yml file though right?

I tried the simple approach, the 3 lines, but I ran into the problem with 'no such file patchelf' when running the example. So I installed patchelf as per the instructions to overcome this, I then receive issues related to import errors (trying to rebuild mujoco_py and 'GLIBCXX_3.4.30' not found).

Thanks, 
Nicos

Vittorio-Caggiano, 2022-09-19:
you won't need to install run the `env.yml` file if you installed myosuite with `pip install -U myosuite`. The `env.yml` file is needed only from the installation from source.

Vittorio-Caggiano, 2022-09-19:
Let me summarize the options to install myosuite in linux.

**Option A - installing directly the pip package**: As described in the [ReadMe](https://github.com/facebookresearch/myosuite)
``` bash
conda create --name myosuite python=3.7.1
conda activate myosuite
pip install -U myosuite
```

**Option B - installing linux dependencies:** As described in the [Advanced Installation](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md) and used in the [Colab](https://colab.research.google.com/drive/1U6vo6Q_rPhDaq6oUMV7EAZRm6s0fD1wn?usp=sharing#scrollTo=JBLQujOKkuZU)
``` bash
sudo apt-get install libgl1-mesa-dev libgl1-mesa-glx libglew-dev libosmesa6-dev build-essential libglfw3 patchelf
conda create --name myosuite python=3.7.1
conda activate myosuite
pip install myosuite
```

**Option C - Installing manually MuJoCo:**  As described in the [Advanced Installation](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md) 
``` bash
wget -q https://github.com/deepmind/mujoco/releases/download/2.1.0/mujoco210-linux-x86_64.tar.gz -O mujoco210.tar.gz
mkdir "$HOME/.mujoco"
tar -zxf mujoco210.tar.gz -C "$HOME/.mujoco"
rm mujoco210.tar.gz
export LD_LIBRARY_PATH="<$HOME/.mujoco>/mujoco210/bin:$LD_LIBRARY_PATH"
conda create --name myosuite python=3.7.1
conda activate myosuite
pip install myosuite
```

**Option D - Installing manually MuJoCo and compile:**  As described in the [Advanced Installation](https://github.com/facebookresearch/myosuite/blob/main/setup/README.md) 
``` bash
wget -q https://github.com/deepmind/mujoco/releases/download/2.1.0/mujoco210-linux-x86_64.tar.gz -O mujoco210.tar.gz
mkdir "$HOME/.mujoco"
tar -zxf mujoco210.tar.gz -C "$HOME/.mujoco"
rm mujoco210.tar.gz
export LD_LIBRARY_PATH="<$HOME/.mujoco>/mujoco210/bin:$LD_LIBRARY_PATH"
conda update conda
git clone https://github.com/facebookresearch/myosuite.git
cd myosuite
conda env create -f setup/env.yml
conda activate myosuite-env
pip install -e .
```

**For Testing:**
``` bash
# Skip this for the Option D as the repo should be already cloned
git clone https://github.com/facebookresearch/myosuite.git
cd myosuite
# Test all environments 
python myosuite/tests/test_myo.py
# Test visualization
python myosuite/utils/examine_env.py --env_name myoElbowPose1D6MRandom-v0
```

Vittorio-Caggiano, 2022-09-20:
Closing this issue for now as it relates to `mujoco-py`. Please, let us know if after installing `mujoco-py` you get any issue with `myosuite`

closed 2022-09-20 by Vittorio-Caggiano