MyoHub/myosuite #26: Evaluation for die reorientation

P-Schumacher, 2022-09-19:
I feel that the evaluation criteria for the die reorientation task are a bit restrictive. Some of my policies are able to solve the task, but only before or after the specific time window that would count as a success.

Would it be possible to relax this a bit?

I suggest an episode limit of 200 and measuring a success if the goal is reached for 5 consecutive time steps at any point in the episode.

This preserves the spirit of the task, but is a bit easier.

NaturalGradient, 2022-09-21:
Just chiming in to say that I agree here. The current die reorientation task doesn't seem generally solveable within 50 time-steps based on my experiments.

vikashplus, 2022-09-22:
- We are strongly considering boosting the horizon of the Die task. Stay tuned.
- The goal of the die task is to stabilize the object at the specified goal location. `success if the goal is reached for 5 consecutive time steps at any point` doesn't seem to capture the essence of this task. There are also a few corner cases for this criteria (a) A policy that aggressively spins the object will succeed (b) A policy that stabilizes the object will have no advantage over a policy that throws the object to goal location. (c) Variable horizon length will introduce artifacts in effort calculations.

P-Schumacher, 2022-09-22:
Thank you for the reply. I didn't consider how the variable horizon length affects the effort calculation.
A slightly longer time interval than 5 steps might have prevented the corner cases, but does not solve the effort issue.

Thinking about it, 200 steps might even be slightly too long then. In my experiments, it's very difficult for the policy to stabilize an object in the tight thresholds during the right time window. But it's hard to say.

Vittorio-Caggiano, 2022-10-09:
closed with #29

closed 2022-10-09 by Vittorio-Caggiano