MyoHub/myosuite #32: State inputs for MyoChallenge

P-Schumacher, 2022-10-27:
Is it possible to make more state information available for the myochallenge?

The tasks might be more feasible if we can include basic information that is readily available to humans, like:
- finger joint velocities
- muscle lengths
- muscle velocities
- muscle fibre forces

I am sure other people might have other requests, feel free to reply!

If the state inputs would be part of the obs_dict, everyone could choose for themselves what to include. All of these are already available in the simulator.

vikashplus, 2022-10-27:
MyChallenge organizers are soliciting feedback on extending access to additional input modalities (this request). A few considerations - 
- We would like to ensure that no privileged information is being passed from the observation space to the policies that otherwise isn't physiologically realistic or isn't easily available in real world for assistive applications.
- Feasibility from an implementation viewpoint
~
MyoChallenge Organizers

pablotanoretamales, 2022-10-28:
Hi MyoChallenge team!

We agree that of course the state-space is not biologically plausible and should be changed in the future. However, we don't think that it is in the best interests of the competition to modify the state space so close to the deadline. We have designed and optimized our agent for one month on the current state space and would need to redesign our approach if the state space is augmented.

P-Schumacher, 2022-10-28:
Of course, it is a bit close to change settings now and require this from everyone. Would you still be against the idea if this state change is entirely optional? You could still use your agent as is and would not need to change anything.

P-Schumacher, 2022-10-30:
okay, I see your point. It's probably fairer to just keep everything like it is then. Good luck.

vikashplus, 2022-10-30:
Considering all the points raised here, consistency with phase1, and fairness to all the participant given the short challenge-timeline, lets keep the input modalities unchanged.

The points raised here are quite valid, as long as no privileged unrealistic information is passed to the policies, we should definitely enable different input modality choices in the future challenge versions.

Thanks for the discussions.

closed 2022-10-30 by vikashplus