MyoHub/myosuite #39: Installing Myosuite on MacOS Ventura 13.3.1

Pouyan97, 2023-04-09:
I followed the advanced installation guide to install Myosuite on my 2019 Macbook Pro (intel core i5). The Conda environment was tested with Python 3.7 and Python 3.8. 
I initially used `brew install gcc` but got an error indicating gcc doesn't exist. Therefore I installed `brew install gcc@7`
I use the following command to link gcc and export the variable. 
`ln -sf /usr/local/bin/gcc-7 /usr/local/bin/gcc` 
`export CC=/usr/local/bin/gcc-7`

Finally, I got the following error with some other files also missing such as `limits.h`: 
[image]
How do I go about solving this? 

I tried to bring all the libraries to the local folder with the following symbolic link command but it didn't seem to be the issue.
`sudo ln -s /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/* /usr/local/include/`

Please let me know if you need more information.

Vittorio-Caggiano, 2023-04-09:
I just tested the installation on Venture 13.3.1 + gcc 12 and it worked fine.

gcc --version:

```
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 12.0.0 (clang-1200.0.31.1)
Target: x86_64-apple-darwin22.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

vikashplus, 2023-04-11:
Can you check if `~/.mujoco/mujoco210/bin` path exists and you have mujoco libraries on your system

Pouyan97, 2023-04-11:
I do have the directory available with the libraries shown below:
[image]

Vittorio-Caggiano, 2023-05-20:
@Pouyan97 closing this for now. Please, reopen it if you get more insights in how we can help fix it

closed 2023-05-20 by Vittorio-Caggiano