MyoHub/myosuite #42: Missing xml files in 1.4.0 installation

decomiteA, 2023-04-28:
I tried to update the latest version (1.4.0) of myosuite using `pip install -U myosuite` but when I try to create an environment it indicates that the associated xml file does not exist : 

`OSError: File ../mujoco_env/lib/python3.8/site-packages/myosuite/envs/myo/../../simhive/myo_sim/finger/myo_finger_v0.xml does not exist` 

When browsing the installed folders I cannot find any mjb or xml files for the different models.

Is there anything I'm missing ?

Vittorio-Caggiano, 2023-04-28:
Hi @decomiteA, is only the `myo_finger` model that doesn't work? do the other model work fine?

closed 2023-04-28 by Vittorio-Caggiano

Vittorio-Caggiano, 2023-04-28:
@decomiteA we found the issue and we pushed a new wheel :1.4.2
Please, let us know if you still see this issue and thanks for the feedback!

decomiteA, 2023-04-28:
Hi @Vittorio-Caggiano thanks for the response! 
I can now see the xml files in the assets folder. 

However I still have errors when I do `gym.make` for the finger, elbow and hand models. All seem to point to the folder "simhive" that does not show up in my local installation.

The error messages are : 

For the finger:
`OSError: File ....../mujoco210/lib/python3.8/site-packages/myosuite/envs/myo/../../simhive/myo_sim/finger/myo_finger_v0.xml does not exist`

For the elbow:
`Exception: Failed to load XML file: ...../mujoco210/lib/python3.8/site-packages/myosuite/envs/myo/assets/arm/myo_elbow_1dof6muscles.xml. mj_loadXML error: b"XML Error: Include error: 'XML parse error at line 0, column 0:\nFailed to open file\n'\nElement 'include', line 10, column 5\n"`

For the hand: 
`Exception: Failed to load XML file: ..../mujoco210/lib/python3.8/site-packages/myosuite/envs/myo/assets/hand/myo_hand_pose.xml. mj_loadXML error: b"XML Error: Include error: 'XML parse error at line 0, column 0:\nFailed to open file\n'\nElement 'include', line 10, column 5\n"`

These errors were reproduced by one of my colleagues on a separate computer

Vittorio-Caggiano, 2023-04-28:
weird.. I tested locally and on a colab and both worked. Generated a new wheel (1.4.3). Again it works both locally and on the colab. Can you please give another try? sorry for the hassle

decomiteA, 2023-04-29:
Everything works now! 

Thank you so much for your dedication  !

adityab, 2023-06-16:
Hi there,

I just installed the newest myosuite today (wheel 1.6.1), and I'm getting those very same exceptions. I also do not have any `simhive` folder mentioned in those xml paths.

Vittorio-Caggiano, 2023-06-17:
Hi @adityab 
I checked locally and with a [colab](https://colab.research.google.com/drive/1U6vo6Q_rPhDaq6oUMV7EAZRm6s0fD1wn?usp=sharing) and the v1.6.1 doesn't have this issue. May I ask you to verify you upgrade to 1.6.1 on your local installation?

commit 40b3e7e9 (2023-04-28): fix #42