MyoHub/myosuite #58: Upgradability to 2.1.1 and above to facilitate compatibility with Mujoco for aarch64 (NVidia Jetson boards)?

rejinjohnvarghese, 2023-06-21:
Dear Myosuite team,

We wanted to use Myosuite in our projects on the NVidia Jetson boards. Mujoco supports aarch64 architecture from 2.1.1 onwards. Is there any way for us to use Myosuite with 2.1.1 or above ? 

Regards,

Rejin

rejinjohnvarghese, 2023-06-22:
Hi Vittorio,

When could we expect to have the binding. Could we expect them in the next few weeks/months or would it be longer. Sorry for bothering you, it was just that we had a few master projects for which we wanted to use the Myosuite platform, however, for these students they have to submit their thesis in August/September - so was just curious whether we can expect to have it in July or would it be much later. 

Sorry for pushing so much.

Regards,

Rejin

Vittorio-Caggiano, 2023-06-22:
The plan is to release it in July. 
Also, we can add some testing of the wheels for aarch64

rejinjohnvarghese, 2023-06-23:
Thanks a lot Vittorio. Really appreciate all your team's work.

closed 2023-06-23 by rejinjohnvarghese