MyoHub/myosuite #62: How to reproduce the DepRL result on MyoLeg walking environment?

Beanpow, 2023-06-29:
Hi, thanks for your helpful environment. The deprl result on the myoleg walking task looks truly natural. However, I have reproduced the training process, and the result don't looks natural. Could you give me some insight on how to generate a natural gait?

P-Schumacher, 2023-06-30:
Hey,
would you mind sharing the exact settings file that you used to train? 
It would also be helpful if you could describe/show the resulting behavior.

Best,
Pierre

Beanpow, 2023-07-02:
Hi Pierre,

Thanks for your help.
I just use the default setting config located at `agents/baselines_DEPRL/myoLegWalk.json`, and I start to train using the following command.
```bash
python -m deprl.main baselines_DEPRL/myoLegWalk.json
```

And here is the score figure.
[image]

As for the generated gait, here is the video.

https://github.com/facebookresearch/myosuite/assets/37178760/a19ca23b-8aed-4e38-90b5-172ac52a57de

Best,
Beanpow

P-Schumacher, 2023-07-02:
Hey Beanpow, 
unfortunately the video won't play for me. However, I have trained a policy with the provided settings file and the result is underwhelming. I was able to find a fix, though.

Instead of training with the original `myoLegWalk-v0`, try to register this environment:
```
register_env_with_variants(id='myoLegWalkRandomInit-v0',
        entry_point='myosuite.envs.myo.walk_v0:WalkEnvV0',
        max_episode_steps=1000,
        kwargs={
            'model_path': curr_dir + '/../../simhive/myo_sim/myoleg/myoleg_v0.52(mj120).mjb',
            'normalize_act': True,
            'min_height':0.8,    # minimum center of mass height before reset
            'max_rot':0.8,       # maximum rotation before reset
            'hip_period':100,    # desired periodic hip angle movement
            'reset_type':'random', # none, init, random
            'target_x_vel':0.0,  # desired x velocity in m/s
            'target_y_vel':1.2,  # desired y velocity in m/s
            'target_rot': None   # if None then the initial root pos will be taken, otherwise provide quat
        }
    )
```
You can add this environment registration to your own script or to [this](https://github.com/facebookresearch/myosuite/blob/main/myosuite/envs/myo/__init__.py) file.

Note that I set the reset-type here to `'random'`. This will cause the model to be initialized in a slightly randomized state and also have either the right or the left leg elevated. This should improve the policy you are seeing, I have tested it myself.

There is another tweak we could do, but I have to discuss that with our team first.

Best,
Pierre

closed 2023-07-11 by Vittorio-Caggiano

Beanpow, 2023-07-12:
Hi Pierre,

Thanks for your help.

Best,
Beanpow

nicos1993, 2023-08-15:
Hello all 

I am trying to install the DEP-RL package `python -m pip install deprl` via Anaconda in Windows 10 (without GPU/CUDA) but I receive an error message about conflicting dependencies: 
`ERROR: Cannot install deprl==0.0.3, deprl==0.0.4, deprl==0.0.5, deprl==0.0.6, deprl==0.0.7, deprl==0.0.8, deprl==0.0.9, deprl==0.1.0, deprl==0.1.1, deprl==0.1.2, deprl==0.1.3, deprl==0.1.4, deprl==0.1.5, deprl==0.1.7 and deprl==0.1.8 because these package versions have conflicting dependencies.`
```
The conflict is caused by:
    deprl 0.1.8 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.1.7 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.1.5 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.1.4 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.1.3 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.1.2 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.1.1 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.1.0 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.0.9 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.0.8 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.0.7 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.0.6 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.0.5 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.0.4 depends on jaxlib<0.5.0 and >=0.4.7
    deprl 0.0.3 depends on jaxlib<0.5.0 and >=0.4.7
```

Any ideas what may be causing the conflict?

Are the installation instructions for Linux OS and/or with GPU/Cuda capabilities?

Best wishes, 

Nicos Haralabidis

P-Schumacher, 2023-08-15:
Hi Nicos,
I have seen this error before, but it’s hard to track down as it is environment dependant.

The deprl default installation was tested on ubuntu 20, 22 and the newest macos. `deprl==0.2.0` also drops the jax dependency as it was causing too many installation issues.

Are you using `python==3.9.0`? I will investigate windows support and report back.

nicos1993, 2023-08-15:
Hello Pierre, 

Thanks for reaching out! 

I am actually using python 3.8.17, the installation instructions for MyoSuite state >= 3.8 is okay - https://myosuite.readthedocs.io/en/latest/install.html ... I will create a new environment with 3.9 and explore. 

Would you advise turning to ubuntu and installing CUDA?

Thanks, 
Nicos

nicos1993, 2023-08-15:
Just an update, 

I created a new environment with Python 3.9 and then I was able to successfully install deprl.

Now I am trying to train a policy as per the original post `python -m deprl.main baselines_DEPRL/myoLegWalk.json`, but I receive the following error: 
`raise ValueError('cannot find context for %r' % method) from None
ValueError: cannot find context for 'fork'`

I will investigate. 

Thanks!

Nicos

P-Schumacher, 2023-08-16:
Hi Nico,
I have to admit: I was aware it would likely not work on windows due to differences in available multiprocessing tools across OSs.

Fortunately for you, I created a pre-release that includes linux, windows and macos support by restructuring multiprocessing inside deprl!

It would be a **tremendous** help if you could help me test this pre-release. I don’t have access to a windows machine now, so I could only do rudimentary testing. This *should* also work with python 3.8.

try it with:
`pip install deprl==0.3.0.dev0`

or 

`pip install deprl --pre`

Thank you for your help,
Pierre

nicos1993, 2023-08-16:
Hello Pierre, 

Okay, that's brilliant! I will give it a try and see how it works.

Best wishes, 

Nicos

nicos1993, 2023-08-16:
I can confirm that `pip install deprl==0.3.0.dev0` leads to `python -m deprl.main baselines_DEPRL/myoLegWalk.json` working!

It has gone through one epoch (?) and logged the data. I'll let it run for a few epochs and then try to visualize the policy - I think it is going to take numerous days more to train given I am not using a GPU (terminal output - `Time left:  epoch 0:06:54  total 3 days, 6:15:26`). 

Also, the wandb link provided here https://myosuite.readthedocs.io/en/latest/baselines.html#dep-rl-baseline seems to require different admin rights.

Thanks for your help!
Nicos