MyoHub/myosuite #73: Issue with DepRL Policy Rollout and Agent Stability

ericwang1997, 2023-08-23:
Hello! 

I recently encountered an issue with the DepRL policy while attempting to generate a rollout of the agent's walking pattern. A few weeks ago, when I conducted the initial test, the agent managed to walk stably for over an hour. However, during this week's testing, its performance significantly deteriorated, causing it to fall after just 1-2 steps. Additionally, I observed a change in the initial position of `myoLegWalking-v0`. Previously, it began from a stable, standing position with both feet firmly on the ground, but the current position seems less stable in comparison. I am using the DepRL [tutorial](https://myosuite.readthedocs.io/en/stable/tutorials.html) to obtain the rollout, attached below. 

Were there any updates made in the past few weeks that could potentially result in this issue? Thank you for your time! 

```
import gym
import myosuite
import deprl
env = gym.make('myoLegWalk-v0')
policy = deprl.load_baseline(env)
obs = env.reset()
for i in range(1000):
    env.mj_render()
    action = policy(obs)
    obs, *_ = env.step(action)
env.close()
```

Vittorio-Caggiano, 2023-08-24:
Hi @ericwang1997 , can you  provide a few more info to reproduce the issue? on which version of MyoSuite you found of this issue and on which it was working? On which Operating System?

P-Schumacher, 2023-08-24:
Hi @ericwang1997 !
I think there were minor changes in initial state and/or collision dynamics which might play a role.
Try:
`env = gym.make('myoLegWalk-v0', reset_type='random')`

this should randomize the initial state and get rid of this problem.

ericwang1997, 2023-08-28:
@Vittorio-Caggiano and @P-Schumacher thanks for the suggestion, the DepRL agent is able to walk without falling for an extended duration for some random initializations, but not all. Since the agent would fall immediately using the default initial states (ie. `reset_type="init"`), I recommend adding this to the DepRL documentation. Thanks again for your help 😄