MyoHub/myosuite #76: Problems about SAR tutorial installment

Ming-Start, 2023-09-01:
Hello! I am trying to use SAR tutorial. But when I install stable-baselines3==1.7.0. I run !pip install stable-baselines3==1.7.0.
then there has errors like below.

> Collecting stable-baselines3==1.7.0
  Using cached stable_baselines3-1.7.0-py3-none-any.whl (171 kB)
Collecting gym==0.21 (from stable-baselines3==1.7.0)
  Using cached gym-0.21.0.tar.gz (1.5 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in gym setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Vittorio-Caggiano, 2023-09-01:
Hi @Ming-Start , the issue seems to be the different versions of gym used by myosuite and stable-baselines3 1.7.0. One workaround I used for colabs is to install a later version of stable-baselines i.e. `pip install "stable-baselines3[extra]>=2.0.0a4"`. Not sure if all results are reproducible with this version of stable-baselines. 

@camberg23 might be able to provide more guidance.

camberg23, 2023-09-01:
Hi @Ming-Start, you could also try installing gym separately before installing stable-baselines3, this might solve the issue.

Ming-Start, 2023-09-02:
> Hi @Ming-Start, you could also try installing gym separately before installing stable-baselines3, this might solve the issue.

Thank you so much @camberg23 

> Hi @Ming-Start , the issue seems to be the different versions of gym used by myosuite and stable-baselines3 1.7.0. One workaround I used for colabs is to install a later version of stable-baselines [...]

Hi @Vittorio-Caggiano . SIr, Thank you so much for your detailed suggestions.

closed 2023-09-03 by Ming-Start

Ming-Start, 2023-09-10:
I have found a good way to solve this problem right now .
First, we need to execute the following code
> pip install --upgrade pip wheel==0.38.4 setuptools==65.5.1
and then , we need to execute the code 
> pip install stable-baselines3==1.7.0
it will be normal now