MyoHub/myosuite #77: Update needed for 4b_reflex tutorial

SoeLinPaing, 2023-09-03:
Hello, I just encountered an issue running the tutorial "4b_reflex". 

When I run the jupyter notebook, it gives the error: ` ERROR: myoLegDemo-v0 not found in env registry` and mjb mentioned in the tutorial is not matching with the mjb file in `myosuite/myosuite/simhive/myo_sim/leg/`. 

Although I replaced those with the environments and mjb files available in the repo, it doesn't work. Any advice on how to fix this? Thank you very much.

elladyr, 2023-09-04:
Hi Soe Lin, thank you for your interest in the reflex model. Yes, this is indeed an error in the tutorial. I will fix it and inform you once it is done.

vikashplus, 2023-09-15:
@elladyr  -- Any update on this issue? If resolved, lets add a summary with a commit ID

elladyr, 2023-09-15:
The fixes have been made, and I am waiting for an independent test before merging the fix.

SoeLinPaing, 2023-09-16:
Thank you very much, Chun Kwang! I have tested the program and it is working now. Just one quick question. It keeps falling after three steps as in the video below. What changes should I make to make it walk longer?

https://github.com/MyoHub/myosuite/assets/51859787/49c5c281-351b-420e-a053-ff65d4551efc

elladyr, 2023-09-16:
Hi SoeLin, Could you tell which branch did you pull it from? Also, could you include the params file you are using?

elladyr, 2023-09-16:
Hi Soe Lin, could you pull from the repository again and run it? The latest updates have been merged into the main branch

SoeLinPaing, 2023-09-16:
Thanks, Chun Kwang! The program is working now.

closed 2023-09-16 by SoeLinPaing