MyoHub/myosuite #80: UnregisteredEnv problem in SAR tutorial Step 1.2

Ming-Start, 2023-09-11:
I followed the tutorial and completed many processes, but when I ran the code in step 1.2 ：

`SAR_RL(env_name=f'myoLeg{new_terrain}TerrainWalk-v0', policy_name='SAR-RL', timesteps=2.5e6, 
       seed='0', ica=ica, pca=pca, normalizer=normalizer, phi=.66, syn_nosyn=False)`

the following error occurred：

> ---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File ~/anaconda3/envs/SAR/lib/python3.8/site-packages/gym/envs/registration.py:132, in EnvRegistry.spec(self, path)
    131 try:
--> 132     return self.env_specs[id]
    133 except KeyError:
    134     # Parse the env name and check to see if it matches the non-version
    135     # part of a valid env (could also check the exact number here)

KeyError: 'myoLegStairsTerrainWalk-v0'

During handling of the above exception, another exception occurred:

UnregisteredEnv                           Traceback (most recent call last)
Cell In[25], line 1
----> 1 SAR_RL(env_name=f'myoLeg{new_terrain}TerrainWalk-v0', policy_name='SAR-RL', timesteps=2.5e6, 
      2        seed='0', ica=ica, pca=pca, normalizer=normalizer, phi=.66, syn_nosyn=False)
      3 #HillyTerrainWalk-v0
      4 #myoLegStairTerrainWalk-v0
      5 #SAR_RL(env_name=f'myoLeg{new_terrain}StairsTerrainWalk-v0', policy_name='SAR-RL', timesteps=2.5e6, 
      6 #       seed='0', ica=ica, pca=pca, normalizer=normalizer, phi=.66, syn_nosyn=False)

Cell In[8], line 17, in SAR_RL(env_name, policy_name, timesteps, seed, ica, pca, normalizer, phi, syn_nosyn)
     15     env = SynNoSynWrapper(gym.make(env_name), ica, pca, normalizer, phi)
     16 else:
---> 17     env = SynergyWrapper(gym.make(env_name), ica, pca, normalizer)
     18 env = Monitor(env)
     19 env = DummyVecEnv([lambda: env])

File ~/anaconda3/envs/SAR/lib/python3.8/site-packages/gym/envs/registration.py:156, in make(id, **kwargs)
    155 def make(id, **kwargs):
--> 156     return registry.make(id, **kwargs)

File ~/anaconda3/envs/SAR/lib/python3.8/site-packages/gym/envs/registration.py:100, in EnvRegistry.make(self, path, **kwargs)
     98 else:
     99     logger.info('Making new env: %s', path)
--> 100 spec = self.spec(path)
    101 env = spec.make(**kwargs)
    102 # We used to have people override _reset/_step rather than
    103 # reset/step. Set _gym_disable_underscore_compat = True on
    104 # your environment if you use these methods and don't want
    105 # compatibility code to be invoked.

File ~/anaconda3/envs/SAR/lib/python3.8/site-packages/gym/envs/registration.py:142, in EnvRegistry.spec(self, path)
    140     raise error.DeprecatedEnv('Env {} not found (valid versions include {})'.format(id, matching_envs))
    141 else:
--> 142     raise error.UnregisteredEnv('No registered env with id: {}'.format(id))

UnregisteredEnv: No registered env with id: myoLegStairsTerrainWalk-v0

Vittorio-Caggiano, 2023-09-11:
Hi @Ming-Start , 

which version of MyoSuite did you install? how did you install it?

Directly with 
``` bash 
pip install -U myosuite
``` 
or, by [cloning the repo](https://github.com/MyoHub/myosuite/blob/main/setup/README.md#editable-installation) with
``` bash
git clone --recursive https://github.com/facebookresearch/myosuite.git
```

Ming-Start, 2023-09-11:
I download Myosuite2.0

camberg23, 2023-09-11:
The correct name is `myoLegStairTerrainWalk-v0`, not `myoLegStairsTerrainWalk-v0`. Your code should work once you correct this! I will push an update to the notebook with this error fixed
@Ming-Start

closed 2023-09-11 by Vittorio-Caggiano

Ming-Start, 2023-09-12:
> The correct name is `myoLegStairTerrainWalk-v0`, not `myoLegStairsTerrainWalk-v0`. Your code should work once you correct this! I will push an update to the notebook with this error fixed @Ming-Start

Thank you so much sir

commit 35ff5fb2 (2023-09-11): Merge pull request #81 from camberg23/patch-1

Update SAR tutorial.ipynb -- fix #80