MyoHub/myosuite #91: Suspected presence of actuator push forces

andreh1111, 2023-09-20:
Hi, 
First of all, thank you very much for the amazing work you are doing with MyoSuite; I am trying to adopt it for my research activity.
I am opening this issue because I encountered something I did not expect. Maybe it is a problem, or maybe I misunderstood something.

I was investigating the various data and attributes of the model and, plotting the env.sim.data.actuator_force attribute while MyoHand was posing ✌, I noticed that the actuator force related to muscle _RI5_ resulted in positive values. I was expecting only negative values (pull forces) or zeros. Also, it is not clear to me why the env.sim.model.actuator_forcerange attribute is always equal to [0,0].

The used code is reported below, and the trained policy is attached [victory_policy3_1M.zip](https://github.com/MyoHub/myosuite/files/12671661/victory_policy3_1M.zip), so that the case study can be reproduced.

Thank you

```
from matplotlib import pyplot as plt
from stable_baselines3 import PPO
import numpy as np
import myosuite
import gym

""" LOAD ENVIRONMENT """
env = gym.make('myoHandPoseFixed-v0', normalize_act=False)
env.reset()

""" LOAD POLICY """
policy = "victory_policy3_1M.zip"
pi = PPO.load(policy)

""" TARGET SETTING """
targetAngles = [env.sim.model.jnt_range[0][1],  # pro_sup
                0,                              # deviation
                env.sim.model.jnt_range[2][0],  # flexion
                env.sim.model.jnt_range[3][0],  # cmc_abduction
                env.sim.model.jnt_range[4][0],  # cmc_flexion
                env.sim.model.jnt_range[5][0],  # mp_flexion
                env.sim.model.jnt_range[6][0],  # ip_flexion
                env.sim.model.jnt_range[7][0],  # mcp2_flexion
                np.mean(env.sim.model.jnt_range[8]),  # mcp2_abduction
                env.sim.model.jnt_range[9][0],  # pm2_flexion
                env.sim.model.jnt_range[10][0], # md2_flexion
                env.sim.model.jnt_range[11][0], # mcp3_flexion 
                env.sim.model.jnt_range[12][0], # mcp3_abduction
                env.sim.model.jnt_range[13][0], # pm3_flexion
                env.sim.model.jnt_range[14][0], # md3_flexion
                env.sim.model.jnt_range[15][1], # mcp4_flexion
                env.sim.model.jnt_range[16][1], # mcp4_abduction
                env.sim.model.jnt_range[17][1], # pm4_flexion
                env.sim.model.jnt_range[18][1], # md4_flexion
                env.sim.model.jnt_range[19][1], # mcp5_flexion
                env.sim.model.jnt_range[20][1], # mcp5_abduction
                env.sim.model.jnt_range[21][1], # pm5_flexion
                env.sim.model.jnt_range[22][1]  # md5_flexion
]
env.env.target_jnt_value = np.array(targetAngles)
env.env.target_type = 'fixed'
env.env.update_target()

""" RENDERING OPTIONS """
env.mj_render()
# set minimal rendering
env.sim.renderer._window.opt.flags[0:-1] = 0
env.sim.renderer._window.opt.flags[[4,7,21]] = 1 # keep only actuator (4), tendon (7), static body (21)
env.sim.renderer._window.opt.geomgroup[1:-1] = 0 # keep only the skeleton
env.sim.renderer._window.opt.sitegroup[0:-1] = 0 # remove sites
env.sim.renderer._window.scn.flags[0:-1] = 0 # remove shadows, reflections, etc.
# camera orientation
env.sim.renderer._window.cam.distance = 0.67
env.sim.renderer._window.cam.azimuth = 90
env.sim.renderer._window.cam.elevation = -10.24
# model position within the screen
env.sim.renderer._window.cam.lookat[0] = -0.1266
env.sim.renderer._window.cam.lookat[1] = -0.1843
env.sim.renderer._window.cam.lookat[2] = 1.364

""" MAIN LOOP """
actuator_force = np.zeros_like(env.sim.data.actuator_force)
for _ in range(3):
    env.reset()
    done = False
    all_rwd = []
    ii = 0
    while not done or ii<100:
        o = env.get_obs()
        a = pi.predict(o)[0]
        next_o, r, done, info = env.step(a)
        actuator_force = np.vstack((actuator_force, env.sim.data.actuator_force))
        ii += 1
env.close()

""" DATA PLOTTING """
musclesList = ['ECRL', 'ECRB', 'ECU', 'FCR', 'FCU', 'PL', 'PT', 'PQ', 'FDS5', 'FDS4', 'FDS3', 'FDS2', 'FDP5',
               'FDP4', 'FDP3', 'FDP2', 'EDC5', 'EDC4', 'EDC3', 'EDC2', 'EDM', 'EIP', 'EPL', 'EPB', 'FPL', 'APL',
               'OP', 'RI2', 'LU_RB2', 'UI_UB2', 'RI3', 'LU_RB3', 'UI_UB3', 'RI4', 'LU_RB4', 'UI_UB4', 'RI5', 'LU_RB5', 'UI_UB5']
# fig, ax = plt.subplots(); ax.imshow(actuator_force.T, aspect='auto'); ax.set_title('actuator_force'); ax.set_xlabel('timesteps'); ax.set_yticks(np.arange(len(musclesList)), labels=musclesList); fig.show()
fig, ax = plt.subplots(); ax.imshow((actuator_force.T>0), aspect='auto'); ax.set_title('actuator_force'); ax.set_xlabel('timesteps'); ax.set_yticks(np.arange(len(musclesList)), labels=musclesList); fig.show()
input()
```

vikashplus, 2023-09-21:
I tried something like this at my end. I can confirm that I see occasional +ve signals in quite a few of the models. I'll need to dig deeper. Will keep this thread updated as I make progress. 

Thanks for flagging this @andreh1111

andreh1111, 2024-01-10:
Hi @vikashplus, I jumped back on this issue because I have often noticed positive forces on muscle _RI5_. Looking at the _Force-Length-Velocity_ (FLV), in particular to the muscle active force (i.e., muscle gain), the problem lies in the _length curve_ contribution (FL). Indeed, gain is computed as -F0 * FL * FV and is expected to be negative, but FL often results in negative values for _RI5_, so the result becomes positive. Below is a simple example in which ctrl is set equal to 1 for _FDS5_ and _RI5_ in a short 500-step simulation.
```python3
import matplotlib.pyplot as plt
import numpy as np
import mujoco

model = mujoco.MjModel.from_xml_path('<your_path>/myohand.xml')
muscle_names = [model.actuator(i).name for i in range(model.nu)]
data = mujoco.MjData(model)
data.ctrl[muscle_names.index('FDS5')] = 1
data.ctrl[muscle_names.index('RI5')] = 1

N = 500
all_actuator_force = np.zeros((N, model.nu))
all_actuator_length = np.zeros((N, model.nu))
all_actuator_gain = np.zeros((N, model.nu))
all_actuator_bias = np.zeros((N, model.nu))

for n in range(N):
    mujoco.mj_step(model, data)
    all_actuator_force[n,:] = data.actuator_force
    all_actuator_length[n,:] = data.actuator_length
    for i in range(model.nu):
        length = data.actuator_length[i]
        lengthrange = model.actuator_lengthrange[i]
        velocity = data.actuator_velocity[i]
        acc0 = model.actuator_acc0[i]
        prmb = model.actuator_biasprm[i,:9]
        prmg = model.actuator_gainprm[i,:9]
        all_actuator_gain[n,i] = mujoco.mju_muscleGain(length, velocity, lengthrange, acc0, prmg)
        all_actuator_bias[n,i] = mujoco.mju_muscleBias(length, lengthrange, acc0, prmb)

fig, axs = plt.subplots(5, 8, figsize=(16, 10))
axs = axs.flatten()
for i in range(len(muscle_names)):
    ax = axs[i]
    line1, = ax.plot(all_actuator_force[:,i])
    ax.set_title(muscle_names[i])
legend_ax = axs[len(muscle_names)]
legend_ax.axis('off')
labels = ['actuator force']
legend_ax.legend([line1], labels, loc='center')
plt.tight_layout()
plt.show()

fig, axs = plt.subplots(5, 8, figsize=(16, 10))
axs = axs.flatten()
for i in range(len(muscle_names)):
    ax = axs[i]
    line1, = ax.plot(all_actuator_gain[:,i])
    line2, = ax.plot(all_actuator_bias[:,i])
    ax.set_title(muscle_names[i])
legend_ax = axs[len(muscle_names)]
legend_ax.axis('off')
labels = ['gain', 'bias']
legend_ax.legend([line1,line2], labels, loc='center')
plt.tight_layout()
plt.show()
```
As you can see from the first plot there are multiple peaks of positive force for _RI5_. From the second plot you can see that active muscle force is responsible. Looking deeper the computation of FL (https://github.com/google-deepmind/mujoco/blob/42515033a7ffcc68f8cb45faba7714910e99ffee/src/engine/engine_util_misc.c#L487), I think I have a clue to the source of the error. In the code below I simply copied the computation of FL for _RI5_ when a positive actuator force is present.
```python3
n_positive, _ = np.where(all_actuator_force > 0)
idx_RI5 = muscle_names.index('RI5')
length_RI5 = all_actuator_length[n_positive[0],idx_RI5]
lengthrange_RI5 = model.actuator_lengthrange[idx_RI5]
range_RI5 = model.actuator_gainprm[idx_RI5][:2]
lmin_RI5 = model.actuator_gainprm[idx_RI5][4]
lmax_RI5 = model.actuator_gainprm[idx_RI5][5]

a = 0.5*(lmin_RI5+1)
b = 0.5*(1+lmax_RI5)
L0 = (lengthrange_RI5[1]-lengthrange_RI5[0]) / (range_RI5[1]-range_RI5[0])
L = range_RI5[0] + (length_RI5-lengthrange_RI5[0]) / L0

FL = 0
if (L >= lmin_RI5 and L <= a):
    x = (L-lmin_RI5) / (a-lmin_RI5)
    FL = 0.5*x*x
elif (L <= 1):
    x = (1-L) / (1-a)
    FL = 1 - 0.5*x*x
elif (L <= b):
    x = (L-1) / (b-1)
    FL = 1 - 0.5*x*x
elif (L <= lmax_RI5):
    x = (lmax_RI5-L) / (lmax_RI5-b)
    FL = 0.5*x*x

print(f'Actuator length = {length_RI5}, length range = {lengthrange_RI5}, L = {L}, FL = {FL}')
```
You can see that the actuator length is smaller than the minimum value of the length range and L is even negative, so the second _else if_ of the sequence is the chosen one. However, the _else if_ sequence present in the computation of FL does not take the eventuality of having length outside the length range into account. In fact, in my opinion, the second _else if_ (line 492) should be ```L > a && L <= 1```, and the third (line 495) ```L > 1 && L <= b```, so that if the length was out of the range, FL would be 0.
Let me know what you think about it.

P-Schumacher, 2024-01-10:
Wow, thanks for this great investigation!

we also occasionally encountered positive forces in our own experiments, also when the muscle length is outside the proper ranges. But we didn‘t trace it back to the FL yet, good find!

These lengthrange violations can always happen in MuJoCo, as all constraints are considered soft. 

The if-else clause is however something that should imho be fixed, it looks like a straightforward bug.
 I personally think a PR to mujoco itself would be appropriate. 
@Vittorio-Caggiano @vikashplus 
Opinions?

vikashplus, 2024-01-10:
I haven't tried the code but if I am following the logic properly this seems sensible. Perhaps @P-Schumacher can double-check the calculation and we can file a PR in MuJoCo. This should be fixed at the MuJoCo level. We can CC this thread in that PR

P-Schumacher, 2024-01-10:
I‘ll double check the fix in the next few days and update here.

i know the function, so it shouldn‘t be too much work

P-Schumacher, 2024-01-16:
Hey @andreh1111  I will look at this more in detail today and tomorrow. Are you compiling mujoco with the updated function? Have you gathered any insights that could help me check this?

P-Schumacher, 2024-01-16:
Indeed, the issue is when a muscles compresses beyond the minimum length and the FL muscleGain function runs into a bug in the if-else clause. Really good find!

I plotted the active part of the FL with the FL in blue, the maximum/minimum muscle length ranges in dashed black and the actually achieved muscle lengths in red
[image]

We can clearly see that the muscle length relationship becomes negative, which forces the muscle to produce lengthening forces, which it should not be able to do.

Fixing the if-else clause with the first two lines here in the mujoco muscleGain function:
```
 if L < lmin_RI5:
     return 0
 elif (L >= lmin_RI5 and L <= a):
     x = (L-lmin_RI5) / (a-lmin_RI5)
     return 0.5*x*x
 elif (L <= 1):
     x = (1-L) / (1-a)
     return 1 - 0.5*x*x
 elif (L <= b):
      x = (L-1) / (b-1)
      return 1 - 0.5*x*x
 elif (L <= lmax_RI5):
      x = (lmax_RI5-L) / (lmax_RI5-b)
      return 0.5*x*x
```
fixes the issue.
See:
[image]

This also prevents the jittering of muscle length issue that has been happening in some muscle configurations. I.e. even for constant activation on all muscles, we got pattern like in the video on the right, which are now fixed on the left.
For this video, I implemented the fix and recompiled mujoco before I set the FDS5 and RI5 muscles to excitations=1.
https://github.com/MyoHub/myosuite/assets/24903880/5c5e0277-463f-4662-8e30-dc79c79df12b

I will create a PR for mujoco itself.

P-Schumacher, 2024-01-17:
The PR has been accepted by the MuJoCo team and committed to their GitHub repo. The next MuJoCo release will contain the bugfix.

closed 2024-01-17 by P-Schumacher