MyoHub/myosuite #116: Help request to apply inverse dynamics on myohand model

andreh1111, 2023-11-13:
Hi,
I'm trying to exploit MuJoCo's inverse dynamics to get `ctrl` given target `qpos`. Following MuJoCo's documentation, I then implemented the following function:

```python
import scipy.optimize as optimize
from copy import deepcopy
import numpy as np
import mujoco

def get_ctrl(model, data, target_qpos):
    data_copy = deepcopy(data)
    # ---- qfrc computation
    data_copy.qpos = target_qpos.copy()
    mujoco.mj_forward(model, data_copy)
    data_copy.qacc = 0
    mujoco.mj_inverse(model, data_copy)
    qfrc = data_copy.qfrc_inverse.copy()
    # ---- afrc computation
    amnt = data_copy.actuator_moment.copy()
    afrc = np.zeros(model.nu,)
    sign_constr = ({'type':'ineq', 'fun': lambda afrc: -afrc}) # only negative actuator forces allowed
    cost_function = lambda afrc: np.linalg.norm(afrc @ amnt - qfrc)
    result = optimize.minimize(cost_function, afrc, method='COBYLA', constraints=sign_constr)
    actuator_force = result.x
    # ---- act computation
    act = np.zeros(actuator_force.shape)
    for idx_actuator, force in enumerate(actuator_force):
        length = data_copy.actuator_length[idx_actuator]
        lengthrange = model.actuator_lengthrange[idx_actuator]
        acc0 = model.actuator_acc0[idx_actuator]
        prmb = model.actuator_biasprm[idx_actuator][:9]
        prmg = model.actuator_gainprm[idx_actuator][:9]
        bias = mujoco.mju_muscleBias(length, lengthrange, acc0, prmb)
        gain = mujoco.mju_muscleGain(length, 0, lengthrange, acc0, prmg)
        act[idx_actuator] = min(max((force - bias) / gain, 0), 1)
    return act
```

Sometimes this function works quite well (i.e., applying the obtained `ctrl` I'm able to get the target position) but often (most of the time) there are individual `qpos` values wrong by about 90°, while other times the obtained position is completely wrong.
I therefore think that maybe I'm doing something very wrong without realising it. 

Do you have any hints about this?
Thank you very much for your support.

Vittorio-Caggiano, 2023-11-15:
Hi @andreh1111 can you add an example with a few trajectories to test when the wrong `qpos` values are generated?

andreh1111, 2023-11-22:
Thanks for your answer @Vittorio-Caggiano!
I did some modifications to the code and now, given a trajectory (and not a single position as I was mistakenly doing before) I can get a sequence of muscle activations suitable for replicating it.
Since I struggled to find examples on the web (particularly on the use of inverse dynamics with muscle actuators) I provide below my current solution hoping to be helpful and to receive feedback in case of mistakes.
```python
def get_act(model, data, target_qpos):
    data_copy = deepcopy(data)
    data_copy.qpos = target_qpos.copy()
    data_copy.qvel = (target_qpos - data.qpos)/model.opt.timestep
    mujoco.mj_forward(model, data_copy)
    data_copy.qacc = 0
    mujoco.mj_inverse(model, data_copy)
    qfrc = data_copy.qfrc_inverse.copy()
    gain = np.array([])
    bias = np.array([])
    bounds = [] #NOTE: optimization w.r.t. act instead of actuator_force allows for act bounded
    # between 0 and 1, rather than too high actuator_force that would then result in clamped act
    bounds_margin = 0.10 # this is to prevent the solution to change too much w.r.t. the previous state
    for idx_actuator in range(model.nu):
        length = data_copy.actuator_length[idx_actuator]
        lengthrange = model.actuator_lengthrange[idx_actuator]
        velocity = data_copy.actuator_velocity[idx_actuator]
        acc0 = model.actuator_acc0[idx_actuator]
        prmb = model.actuator_biasprm[idx_actuator,:9]
        prmg = model.actuator_gainprm[idx_actuator,:9]
        bias = np.append(bias, mujoco.mju_muscleBias(length, lengthrange, acc0, prmb))
        gain = np.append(gain, mujoco.mju_muscleGain(length, velocity, lengthrange, acc0, prmg))
        bounds.append((max(0, data.act[idx_actuator]-bounds_margin), min(1, data.act[idx_actuator]+bounds_margin)))
    cost_function = lambda act: np.linalg.norm((gain * act + bias) @ data_copy.actuator_moment - qfrc)
    result = optimize.minimize(cost_function, data.act, method='SLSQP', bounds=tuple(bounds))
    return result.x
```
At first my intention was to compute `ctrl` given obtained `act`, i.e.:
```python
''' for each trajectory sample '''
    act = get_act(model, data, target_qpos)
    act_dot = (act - data.act)/model.opt.timestep
    tau = model.actuator_dynprm[:,0] * (0.5 + 1.5 * act)
    tau[act_dot<=0] = model.actuator_dynprm[act_dot<=0,1] / (0.5 + 1.5 * act[act_dot<=0])
    ctrl = act_dot * tau + act
    data.ctrl = ctrl
    mujoco.mj_step(model, data)
```
but I then realized that doing so the `act` resulting from applying `ctrl` would be different from the target `act` computed previously. Not knowing how to efficiently implement the muscle filter dynamics in `get_act`, I then decided to directly set `data.act = get_act(model, data, target_qpos)`, with `model.actuator_dyntype = mujoco.mjtDyn.mjDYN_INTEGRATOR` instead of `mjDYN_MUSCLE`.

Note 1: Since I have not yet implemented any closed-loop feedback in this solution, in case the replication of the target trajectory generates enough contact forces to prevent the movement of a finger of myohand, in the long run the resulting trajectory of that finger will be different from the target one.
Note 2: Although the purpose of this implementation is related to the control of MyoSuite (of myohand in particular), I understand that I may have been wrong in not opening the conversation among MuJoCo issues, and in that case I apologize.

Vittorio-Caggiano, 2023-11-25:
Hi @andreh1111 , I think having this as an example/tutorial could be useful for the community. We could add it to our [tutorials](https://github.com/MyoHub/myosuite/tree/main/docs/source/tutorials) section?

andreh1111, 2023-11-26:
Done, happy to contribute. Thanks again @Vittorio-Caggiano.

closed 2023-11-27 by Vittorio-Caggiano