MyoHub/myosuite #122: How to find the internal implementation code of the task 'myoHandReorient8-v0'

wbella, 2023-12-11:
Hi ,when I run the SAR code, I can not see the internal definition of the 'myoHandReorient8-v0' environment in the myosuite, such as its step() function, reset() function, etc.. Could you tell me where I should go to find the internal implementation code of this environment. Thanks a lot.

Vittorio-Caggiano, 2023-12-11:
Hi @wbella , the definition of the  [myoHandReorient8-v0](https://github.com/MyoHub/myosuite/blob/440525a2a81b96ef5fd2b2918029f26905b78dbc/myosuite/envs/myo/myobase/__init__.py#L480) uses [reorient_sar_v0.py](https://github.com/MyoHub/myosuite/blob/440525a2a81b96ef5fd2b2918029f26905b78dbc/myosuite/envs/myo/myobase/reorient_sar_v0.py#L14)  which inherits those functions from [base_v0.py](https://github.com/MyoHub/myosuite/blob/440525a2a81b96ef5fd2b2918029f26905b78dbc/myosuite/envs/myo/base_v0.py)

wbella, 2023-12-11:
Thanks a lot!

> Hi @wbella , the definition of the [myoHandReorient8-v0](https://github.com/MyoHub/myosuite/blob/440525a2a81b96ef5fd2b2918029f26905b78dbc/myosuite/envs/myo/myobase/__init__.py#L480) uses [...]

closed 2023-12-11 by Vittorio-Caggiano