MyoHub/myosuite #133: Cannot reproduce SAR-RL results in the MyoLeg Hilly Locomotion task

dhruv-sarkar, 2024-01-03:
[image]

This is the performance I am getting when trying the ICAPCA approach with phi = 1 and phi = 0.66. 

It is clearly worse compared to the results published in the paper. To be sure that the worse results are not due to something I did wrong in the play phase,  I took the precomputed SAR values that are provided in this repository.

Below I provide the results given in the paper. As you can see my results are falling significantly short of the ones provided in the paper.

[image]

camberg23, 2024-01-05:
Hi @dhruv-sarkar, a few suggestions for troubleshooting this:

1. Have you tried rendering the policies and seeing if the agent/environment behavior looks reasonable? This could help us determine where the problem is. I personally recall getting results quite similar to this when there was an issue with the contacts in the environment (the agent was stepping 'through' the ground).
2. There will be seed-to-seed variation, consider running them again if 1. doesn't reveal an obvious issue.
3. Try running the precomputed SAR on a different related environment and see if it still isn't working.

I can try to help again once you've given 1-3 a try if the issue is still not resolved at that point. Miscellaneous note: the environments may have changed slightly since the paper—consider getting SAR from scratch if this is indeed the case. Hope this helps.

dhruv-sarkar, 2024-01-05:
Thanks @camberg23 for taking care to write such a detailed reply. 

> 1. Have you tried rendering the policies and seeing if the agent/environment behavior looks reasonable? This could help us determine where the problem is. I personally recall getting results quite [...]

I have tried 1 and yes the agent seems to have a contact problem. What's the next step?

vikashplus, 2024-01-05:
@cameronberg -- If we suspect code regression, can you also point out the commit hash where the results worked at your end. @dhruv-sarkar -- it might be a good idea to check that hash at your end before we go ahead about fixing the latest version.

P-Schumacher, 2024-01-15:
Hi @dhruv-sarkar ! this might have been due to issues with the height field in MuJoCo. We pushed fixes to this to the GitHub repo. See https://github.com/MyoHub/myosuite/commit/bdb973ac655e0596a7040e5d9758e525109117e6. 

Try installing myosuite by cloning from GitHub and please test if the performance improves!
We would be glad to hear back from you.

dhruv-sarkar, 2024-01-18:
Sure. I tried running the experiments on the various terrains with precomputed SAR and these are the various results I got. 
[image]

I think that is reasonable. I am also attaching the videos in case someone is curious.

https://github.com/MyoHub/myosuite/assets/121563009/d2e84739-f498-429e-8150-5b28d8c5427b

https://github.com/MyoHub/myosuite/assets/121563009/b95e04c8-4d0d-4521-ba7a-b0d4698e4f8b

https://github.com/MyoHub/myosuite/assets/121563009/32e11100-5475-43ee-920b-22e7ae9dbba8

So are we good to close this issue?

P-Schumacher, 2024-01-18:
Looks close enough to the original score for me. @Vittorio-Caggiano  can be closed right? I'm not the SAR expert.

Vittorio-Caggiano, 2024-01-18:
@dhruv-sarkar thanks for the update and for following up on this issue. Were those results obtained with the standard tutorial/seed or did you need to make any change?

dhruv-sarkar, 2024-01-18:
@Vittorio-Caggiano I did not need to make any change. In fact I even used the precomputed SAR given in the repo to generate these results. This perhaps shows that even the precomputed SAR that has been provided is still relevant.

Vittorio-Caggiano, 2024-01-18:
sg... we can then close the issue.
thanks again @dhruv-sarkar for following through on this issue!

closed 2024-01-18 by Vittorio-Caggiano

Baliencasia, 2024-07-24:
@dhruv-sarkar  Hello, have u ever tried to reproduce SAR-RL results in the Myohand reorient task?

dhruv-sarkar, 2024-07-24:
> @dhruv-sarkar Hello, have u ever tried to reproduce SAR-RL results in the Myohand reorient task?

@Baliencasia No. Unfortunately, I haven't tried reproducing on Myohand reorient task.