MyoHub/myosuite #136: Change target position and color on Pen Twirl

Vittorio-Caggiano, 2024-01-10:
Pen twirl environments have them follow the same format of other in hand object rotation environments

vikashplus, 2024-03-11:
@Vittorio-Caggiano  -- can you provide more details on what is required here? I'm not sure I follow the suggestion yet.

Vittorio-Caggiano, 2024-03-11:
there is a yellow sphere to indicate the achievement of the goal. This sphere in the pen-twirl task covers the rotated object instead of the target orientation. In other tasks e.g. die-rotation, the yellow sphere is correctly on the target object.
[image]
[image]