MyoHub/myosuite #139: How to generate the videos or snapshots of myosuite tasks with a black background like the figures in the SAR paper?

wbella, 2024-01-20:
Hi, I have verified a new proposed algorithm by using myosuite environment, but when writing a paper, I want to provide snapshots or videos of the movements of the musculoskeletal hand.  The videos I've generated so far show the musculoskeletal arms in rooms with furniture. How can I draw the pictures with a black background like the Fig. A.16 in the SAR paper( SAR: Generalization of Physiological Agility and Dexterity via Synergistic Action Representation)?
Thanks a lot!

vikashplus, 2024-01-20:
For onscreen viewer 
- press `d` to toggle the scene
- press `1` to toggle the skin

Vittorio-Caggiano, 2024-01-21:
Hi @wbella, thanks for using MyoSuite. In addition to Vikash's suggestion, you can also remove the background programmatically with 

``` python
# Find geometries with ID == 0 which include floor and room
geom_0_indices = np.where(env.sim.model.geom_group == 0)
# Change the alpha value to make it transparent
env.sim.model.geom_rgba[geom_0_indices[0][0], 3] = 0 # to make the floor transparent
env.sim.model.geom_rgba[geom_0_indices[0][1], 3] = 0 # to make the room transparent
```

closed 2024-01-21 by Vittorio-Caggiano

wbella, 2024-01-21:
Thanks a lot!