MyoHub/myosuite #140: Addition of the Adductor Pollicis muscle

andreh1111, 2024-01-20:
Hi,
have you ever considered adding the _adductor pollicis_ muscle? I think it would be very useful if MyoHand had it to simplify/improve thumb dexterity.

Vittorio-Caggiano, 2024-01-21:
Hi @andreh1111 , we are considering extending the thumb muscles to have more expressive behaviors. Do you have any benchmark on specific movements that are not possible to achieve? We could use those for some testing.

andreh1111, 2024-01-21:
I have no data on specific impossible movements, partly because the palmar adduction of the thumb is currently achievable with the proper co-contraction of _EPL_ and _FPL_, but the presence of _adductor pollicis_ would allow a more physiological behavior.

Doing some tests with inverse dynamics however I encountered some challenges in meeting the target position for all thumb joints. Maybe with reinforcement learning it is possible to get better results, or maybe not (but I don't want to start a competition about it in this thread, I'm too scared of what @P-Schumacher can do). Joking aside, if even with RL a trajectory like the one used below is hard to meet for all thumb joints, perhaps it can be used as a first benchmark.

In the gif below (github complained with the video, sorry) you can see the movement I generated as an example (trajectory samples in the [attached](https://github.com/MyoHub/myosuite/files/14002812/issue140_trajectory.csv) .csv). On the left is the reference, on the right what was obtained using inverse dynamics. In the first image you can see the reference vs achieved trajectory (note: as indicated in #112, the palmar abduction/adduction of the thumb is currently titled _cmc_flexion_ instead of _cmc_abduction_). There is a drift mainly at the expense of _mp_flexion_ and _ip_flexion_, in my opinion due to the use of _EPL_ and _FPL_ to achieve the desired trajectory. The _ctrl_ obtained for the muscles of the thumb is reported in the last image.

[image]

[image]

[image]