MyoHub/myosuite #145: mplementing a torque sensor in myoElbowPose1D6MRandom-v0

MaximilianKinseher, 2024-02-06:
Hi, 
I want to implement a torque sensor to the elbow joint of the 'myoElbowPose1D6MRandom-v0' environment. 
My problem is I am unable to find the code for this environment. 
Thanks in advance!

Vittorio-Caggiano, 2024-02-06:
Hi @MaximilianKinseher , 
the model used to build this task is the `myoElbowPose1D` which you can find at `myosuite/simhive/myo_sim/elbow//myoelbow_1dof6muscles.xml` 

... do not forget to clone with recursive option: `git clone --recursive https://github.com/facebookresearch/myosuite.git` -- see [here](https://myosuite.readthedocs.io/en/latest/install.html#alternative-installing-from-source)

closed 2024-02-06 by Vittorio-Caggiano

MaximilianKinseher, 2024-02-07:
Thank You @Vittorio-Caggiano, 
If I edit this myosuite/simhive/myo_sim/elbow//myoelbow_1dof6muscles.xml file in my forked version of myosuite 
and use !pip install git+https://github.com/MaximilianKinseher/myosuite in my colab notebook, 
will my changes be implemente when I use env = gym.make('myoElbowPose1D6MRandom-v0')?
Or do I have to create a new environment in gym with my edited version of myosuite/simhive/myo_sim/elbow//myoelbow_1dof6muscles.xml? 
Thanks in advance!