MyoHub/myosuite #153: Simulating musculoskeletal model using mocap data in myosuite

SuryaPratapSingh37, 2024-03-06:
Hi,

Firstly, kudos for your amazing work! I'm curious how can I transfer marker/keypoints-based real-world motion capture data to MyoSuite for simulating musculoskeletal models, similar to the process in OpenSim?

Additionally, how can I command the joints of these models to follow specific trajectories. Do you have any reference codes or guidance on achieving this?

Vittorio-Caggiano, 2024-03-11:
Hi @SuryaPratapSingh37 
thanks for your support!

In order to achieve that transfer you will need to add the markerset ([mocap bodies](https://mujoco.readthedocs.io/en/latest/modeling.html#mocap-bodies)) specific for your motion capture system to the muskuloskeletal models and from there use [Inverse Kinematics](https://github.com/google-deepmind/dm_control/blob/main/dm_control/utils/inverse_kinematics.py) potentially together with other optimizations (see https://arxiv.org/html/2402.17192v1).

We are planning to release more tools to support those functionalities soon

connected issue to: https://github.com/MyoHub/myo_sim/issues/21#issue-2167878056

Kavya-mothukuri, 2025-01-28:
> Hi @SuryaPratapSingh37 thanks for your support!
> 
> In order to achieve that transfer you will need to add the markerset ([mocap bodies](https://mujoco.readthedocs.io/en/latest/modeling.html#mocap-bodies)) specific for your motion capture system to the muskuloskeletal models and from there use [Inverse Kinematics](https://github.com/google-deepmind/dm_control/blob/main/dm_control/utils/inverse_kinematics.py) potentially together with other optimizations (see https://arxiv.org/html/2402.17192v1).
> 
> We are planning to release more tools to support those functionalities soon
> 
> connected issue to: [MyoHub/myo_sim#21 (comment)](https://github.com/MyoHub/myo_sim/issues/21#issue-2167878056)

I'm having the same doubt. Can you provide any tutorial for this?