MyoHub/myosuite #156: Reaching Arm model

LyesBesylex, 2024-03-16:
Any idea on how I can alter reachv0 for the arm model like you did for myoLegStandRandom-v0?

Vittorio-Caggiano, 2024-03-18:
Hi @LyesBesylex what are you exactly trying to do? In which way do you want to modify the reach environment?

LyesBesylex, 2024-03-18:
Hello @Vittorio-Caggiano,
I want to create a new environment where the MyoArm model will specifically target the object and move the arm to grasp it. This process is done in relocateV0 but it also moves the object to a targeted bin. I only want the arm to reach an object.

Vittorio-Caggiano, 2024-03-25:
hi @LyesBesylex , added [here](https://github.com/MyoHub/myosuite/blob/dev/docs/source/tutorials.rst#customizing-tasks) some instructions. Please let me know if this is what you are looking for

closed 2024-03-25 by Vittorio-Caggiano