MyoHub/myosuite #174: Issue with Simulation of Hand Environment

ziorex, 2024-05-23:
Hello,

I am currently working through the Inverse Dynamics tutorial provided in the MyoSuite repository. I extracted the muscle control data from the provided tutorial "6_Inverse_Dynamics.ipynb", which saves also the myohand_freemovement.mp4 video I attach.
https://github.com/MyoHub/myosuite/assets/146551963/c16de44a-994f-48ff-9960-e360faa88b3f

Subsequently, I loaded the actuator activation data into another script and used it to run a simulation of one of the environments provided by the library (I attach the "script.py" I used for it and the "simulation.mp4" I recorded).
https://github.com/MyoHub/myosuite/assets/146551963/fc324209-c575-4a4c-bb64-cf6170a437d9
**Script:**
`
import myosuite
import gym
from tqdm import tqdm
import numpy as np
import os

env = gym.make("myoHandPoseFixed-v0")
obs = env.reset() 

all_ctrl = np.load(os.path.join(os.path.dirname(__file__), "all_ctrl_standard.npy"))
print(all_ctrl)

for idx in tqdm(range(all_ctrl.shape[0])):
    env.step(all_ctrl[idx, 1:])
    env.sim.renderer.render_to_window()
`

**Problem:**
The simulation appears to be influenced by some type of constraint or limit that affects the trajectory.
Could you please provide some guidance on why this might be happening and how to resolve this issue?

Thank you for your assistance.

**Info software version:**
- myosuite = 2.2.0
- mujoco = 2.3.7
- gym = 0.13.0
- python = 3.11.0

andreh1111, 2024-05-23:
Hi @ziorex, thanks for your interest in the tutorial.

In this tutorial, muscle activations are found by solving a quadratic program for which the _tausmooth_ parameter must be set (you can check [here](https://mujoco.readthedocs.io/en/latest/modeling.html#muscles) for its meaning), and the same value must be set when those muscle activations are reused.

So you need to update your code before the for loop with:
`env.unwrapped.sim.model.actuator_dynprm[:,2] = 5`

Let me know if this fixes the problem.

ziorex, 2024-05-23:
Unfortunately this is not solving the issue, I think it is a problem related with the environment itself, since even if I run a simulation with random activations I see the model's muscles activated randomly but with the hand being constrained to the same position of before.
https://github.com/MyoHub/myosuite/assets/146551963/38fb4673-f6d5-4d87-a20f-00c2a3287196

Thank you so much for your tempestive answer!

andreh1111, 2024-05-23:
When you create the environment you need to do:
`env = gym.make("myoHandPoseFixed-v0", frame_skip=1, normalize_act=False)`

The `frame_skip` flag by default is equal to 10 (timestep=0.02) but inverse dynamics computed the ctrls step by step (timestep=0.002).

The `normalize_act` flag by default is True to normalize actions (useful for RL) but this normalization is not used by inverse dynamics.

These updates all together should solve the issue, I tested on my computer and everything works fine, let me know from your side.

ziorex, 2024-05-26:
The simulation is definitely better with the adjustment you've given, but it is still influenced by a kind of constraint or limit which disturbate the simulation that should be equal to the video coming from the inverse dynamic tutorial I shared in the first message.
The following is the video of the simulation with the script updated:
https://github.com/MyoHub/myosuite/assets/146551963/42b695aa-ae2c-493c-bc04-14ab8b389603

script:
import myosuite
import gym
import numpy as np
import os
import time
from tqdm import tqdm

env = gym.make("myoHandPoseFixed-v0", frame_skip=1, normalize_act=False)
all_ctrl = np.load(os.path.join(os.path.dirname(__file__), "all_ctrl_standard.npy"))
env.unwrapped.sim.model.actuator_dynprm[:,2] = 5
obs = env.reset() 

for idx in tqdm(range(all_ctrl.shape[0])):
    env.step(all_ctrl[idx, 1:])
    env.sim.renderer.render_to_window()

andreh1111, 2024-05-26:
I was able to replicate an issue similar to yours by generating ctrls with mujoco 3.1.2 and using them with mujoco 2.3.7. Maybe you generated the ctrls using the tutorial on Google Colab, and then tested them on your computer? If this is the case all you need is a myosuite update.

ziorex, 2024-05-27:
Now the trajectory is followed correctly, the error it was effectively in the calculation of the ctrls with the old version of the tutorial of inverse dynamics; so that, combining the right ctrls with the adjustment you recommended in the previous answers I was able to solve the issue. Thank you very much.