MyoHub/myosuite #175: Initial robot pose and initial object pose not available.

anjugopinath, 2024-05-29:
'robot_init',  # shape(n_objects_jnt) ==> initial robot pose
 'object_init'  # shape(n_objects_jnt) ==> initial object

I am referring to this file : https://github.com/MyoHub/myosuite/blob/main/myosuite/envs/myo/myodm/data/post_process.py

When I execute it, I get 'Key Error' for the above 2 fields.

vikashplus, 2024-05-31:
There was a bug that I thought I fixed but maybe not. There is a typo in the keyname. `robot_int` ==> `robot_init`
            robot_init=data['robot_int'],
            object_init=data['object_int'],

anjugopinath, 2024-06-16:
Hi Vikash,

Thank You for your response.
I am still getting an error:
[image]

Also,  I wrote some code to print all the keys in the dictionary:

for npz_file in npz_files:
    with np.load(npz_file) as data:
        print(f"Keys in {npz_file}: {list(data.keys())}")

I think only 'time', 'robot' and 'object' are the 3 fields available :
[image]

fl0fischer, 2026-10-05:
In the upcoming 3.0 release (branch `ms3`, PR #406) the MyoDM suite (the `myodm` envs and `post_process.py`) is removed, so this script no longer exists. Closing; the earlier releases still contain it.

closed 2026-10-05 by fl0fischer