MyoHub/myosuite #176: computed muscle control(CMC) in myosuite/mujoco?

Baliencasia, 2024-06-05:
Hello,
Thanks for the great work!
Is it possible to use computed muscle control(CMC), static optimization, or inverse dynamics methods which these methods are widely used in opensim to calculate muscle activation from motion data in myosuite/mujoco? Is there such an API in myosuite, or is there a repo for related functions.
By the way, is there a corresponding OSIM model for [myoarm](https://github.com/MyoHub/myo_sim/blob/main/arm/assets/myoarm_assets.xml) with 63 muscles and 27 DOF?  If so, where can I get it?
Thanks again for the your excellent work.

Vittorio-Caggiano, 2024-06-05:
Hi @Baliencasia , it is definitely possible. @andreh1111 has made a tutorial](https://github.com/MyoHub/myosuite/blob/main/docs/source/tutorials/6_Inverse_Dynamics.ipynb) to show an example of this. In principle it should be possible to use static optimization or other methods. If you have solutions that you would like to share with the community, it would be fantastic if you could contribute them with tutorials.

We do not have the corresponding Osim models. Our process requires optimizations which need to be done directly in MuJoCo. For which use case you would need to use Opensim?

Baliencasia, 2024-06-05:
wow, it's great. I'm sorry I overlooked the tutorial of inverse dynamics before. And if there is inverse dynamics in mujoco, the requirement for the OSIM model may not be needed anymore. Thank you again, your profile photo is so cool. hhhhhh

Vittorio-Caggiano, 2024-06-05:
You can [join our slack](https://join.slack.com/t/myosuite/shared_invite/zt-1zkpw2zzk-NhVhVlSDxhoMHbzROD8gMA) too and let us know how it goes.  Really curious to see the results with those optimization methods :)

closed 2024-06-05 by Vittorio-Caggiano

Baliencasia, 2024-06-05:
got it

scakki, 2024-09-06:
Hello!
Thank you for the great work and resources provided. I am currently attempting to run the [Inverse Dynamics Tutorial](https://github.com/MyoHub/myosuite/blob/main/docs/source/tutorials/6_Inverse_Dynamics.ipynb), but it keeps getting terminated even after consuming 138 GB of RAM and 1.5 TB of swap space. I'm unsure if I am making a mistake or if the tutorial has exceptionally high RAM requirements. Could you please provide information on the system requirements to run this code successfully?

scakki, 2024-09-06:
Thank you for the response. Myosuite version is 2.7.0, python 3.9.19, and regarding the data ('data/6_trajectory.csv') and model (myohand), they are the default ones specified in the tutorial. Apologies, I don't have a minimal example.

andreh1111, 2024-09-07:
Hi @scakki, most likely the issue is due to deepcopy of MjData, which for some versions of Mujoco results in memory leaks (see [here](https://github.com/google-deepmind/mujoco/issues/1572)). The bug was fixed with the Mujoco 3.2 release, so you should just need a Mujoco upgrade. Let me know if this fixes the issue on your side.

Baliencasia, 2024-09-07:
hi @scakki, in my use, IK does not require a large RAM and computing resources, and a regular computer can complete inverse dynamics calculations in a few minutes, so maybe you need a Mujoco upgrade as mentioned above.

scakki, 2024-09-10:
Thank you @andreh1111 and @siyuan-liu-casia, upgrading MuJoco version fixed the issue.

Vittorio-Caggiano, 2024-09-10:
thanks @scakki and @andreh1111 to troubleshoot this issue!