MyoHub/myosuite #179: Visualize trajectory step by step

anjugopinath, 2024-06-16:
Hello,

I want to visualize the trajectory of the robot and the object in unison.
I am using the .npz files in this path : https://github.com/MyoHub/myosuite/tree/main/myosuite/envs/myo/myodm/data

[image]

Is this something I can use?
https://github.com/MyoHub/myosuite/blob/main/docs/source/tutorials/6_Inverse_Dynamics.ipynb

Vittorio-Caggiano, 2024-08-06:
@anjugopinath it seems you can access the trajectories of the object and hand. I do not understand the inverse dynamics question though. Can you clarify what are you trying to do ?

justinthecao, 2024-11-19:
What do the 29 dof represent for the robot? Also what do the 7 values represent for the object.

Vittorio-Caggiano, 2024-11-19:
23 degrees of the hand + translation/rotation (6 dof) of the shoulder/arm

justinthecao, 2024-11-19:
so it would be [0....22] for joint angles
[23...25] for XYZ robot position
[26...28] for robot rotation?

justinthecao, 2024-11-19:
Yeah, I saw that function, but I can't seem to find where the robot values represent (it just copies the robot ref). I specifically want the positions of the robot hand.

Vittorio-Caggiano, 2024-11-19:
you were correct, the first 23 elements correspond hand joints

justinthecao, 2024-11-19:
How about the translation? Is it the 3 elements after that?

Vittorio-Caggiano, 2024-11-21:
yes, first 3 after are translation and then orientation
https://github.com/MyoHub/myosuite/blob/7927eb0153b428812c1db143e0e41144ab57254b/myosuite/envs/myo/myodm/myodm_v0.py#L315