MyoHub/myosuite #181: A question about maintaining contact between the human model and some rigid bodies

Yingfan99327, 2024-06-28:
Hi,

May I ask a question about how to maintain the contact between a myosuite model and a rigid body defined by the user in mujoco?

For example, if I load the myohand_pen.xml and myohand_baoding.xml into the mujoco.viewer, the pen or the ball don't maintain a consistent contact with the human hand as shown in the picture below. And, I can also see that there are two tendons connecting the human hand and the two baoding balls. However, from the released demos, I can see that the two baoding balls and the pen are actually played on the hand model.

So, I am wondering how to maintain such a consistent contact between a hand model and a rigid body. I have two guesses,

1) It should be defined in the original xml model file, but I am still not clear about the function of the two tendons connecting human hand and baoding balls in the myohand_baoding.xml;
2) Maintaining the contact between human hand model and a rigid body is also a part of the tasks which should be defined in the corresponding env.py file.

Thanks very much and wishing you all the best.[](url)
[image]
[image]

Vittorio-Caggiano, 2024-07-01:
Hi @Yingfan99327 I think those are 2 different points. 
1. When you play the environment with simulate, you might not be actuating the hand. In order to maintain contact between hand and object, you need to activate the hand muscles in order to move the fingers to grab and hold on the object. You can check this out playing back the agents policies here: https://github.com/MyoHub/myosuite/tree/main/myosuite/agents 
2. in the baoding ball task, those two tendons are connecting the current position of the object and the target position. You can see the definition here: https://github.com/MyoHub/myosuite/blob/1c02bdb66fc5c154bc332b6068203cc0cb7239e0/myosuite/envs/myo/assets/hand/myohand_baoding.xml#L42:L51

Yingfan99327, 2024-07-03:
Really appeciate your reply! Thanks very much and wishing you all the best.

closed 2024-07-03 by Yingfan99327