MyoHub/myosuite #182: Mummified Musculoskeletal Model

irfanmohamed12, 2024-07-05:
Executing a simple step of the environment results in a weird reproduction of the musculoskeletal model. 

env = gym.make('myoElbowPose1D6MRandom-v0')
env.reset()
for _ in range(1000):
  env.mj_render()
  env.step(env.action_space.sample()) # take a random action
env.close()

[image]

Any reason why this issue occurs?

Vittorio-Caggiano, 2024-07-05:
hi @irfanmohamed12 , can you provide operating system, python version and myosuite version?

Vittorio-Caggiano, 2024-08-06:
@irfanmohamed12 is this issue still open?

irfanmohamed12, 2024-08-07:
it is still open indeed...

Vittorio-Caggiano, 2024-08-07:
@elladyr can you help replicating this issue?

elladyr, 2024-08-11:
@Vittorio-Caggiano 
I managed to replicate this issue on a few combinations of platform and Myosuite versions
- Myosuite versions
-     2.4.0
-     2.5.0
- Win 11 Pro
-     Python 3.9.13
-     Python 3.10.14
- Win 11 WSL (Ubuntu 20.04)
-     Python 3.9.13
-     Python 3.10.14
- Render modes
-     Onscreen rendering - mj_render(), env.renderer.render_to_window()
-     Offscreen rendering - env.renderer.render_offscreen()

When rendering, it seems that the convex hull flag is always activated. This affects all the myoElbow and myoElbowExo environments. I have also checked on the Locomotion environments and they are unaffected.

How should we proceed in this case?

PeterHe0226, 2024-08-15:
To solve the issue, unwrap the linen on the mummy manually. First `import Numpy,` then use `np.unwrap()` until the body is exposed.

Vittorio-Caggiano, 2024-08-16:
thanks @PeterHe0226, really appreciate your help!

Can you please add a snippet of code how this fix should run?

vikashplus, 2024-09-04:
I just tested it on Mac with `MyoSuite version: 2.5.0` and it looks perfect. So maybe the issue is something to do with windows. Can someone check with windows machine check the following command on Windows and report back - 

`python myosuite/utils/examine_env.py -e myoElbowPose1D6MRandom-v0`

[image]

vikashplus, 2024-09-04:
@elladyr - can you try versions I posted above on windows so we can rule out platforms as the issue as well.

elladyr, 2024-09-04:
@vikashplus

I checked the version already and it is a specific  Windows problem. When rendering, it seems that the convex hull flag is always activated. This affects all the myoElbow and myoElbowExo environments. I have also checked on the Locomotion environments and they are unaffected.

How should we proceed in this case?

Tested on:
Myosuite versions

- 2.4.0
- 2.5.0

Win 11 Pro

- Python 3.9.13
- Python 3.10.14

Win 11 WSL (Ubuntu 20.04)

- Python 3.9.13
- Python 3.10.14

Render modes

- Onscreen rendering - mj_render(), env.renderer.render_to_window()
- Offscreen rendering - env.renderer.render_offscreen()

vikashplus, 2024-09-04:
@elladyr 
Can you try the latest MyoSuite with these versions of python, pip, and numpy 
```
python 3.8.18 
pip 23.3.1
numpy 1.24.4
```

elladyr, 2024-09-04:
Checked with both Win 11 Pro and Win 11 WSL (Ubuntu 20.04).
Checked for Myosuite versions
- 2.4.0
- 2.5.0

using

`python 3.8.18 
pip 23.3.1
numpy 1.24.4`

No convex hull activated
[image]

vikashplus, 2024-09-05:
That confirms that it's a dependency issue.

Vittorio-Caggiano, 2024-09-05:
this seems to be a MuJoCo issue about not supporting yet `numpy> 2.0` https://github.com/google-deepmind/mujoco/issues/1808#issuecomment-2328732267

vikashplus, 2024-09-05:
Should we pin the Python version and numpy version in that case?

gandie, 2024-09-07:
Can confirm this issue on Windows using Python 3.12 .

Workaround was to install yet another Python Interpreter ( 3.8 , im a collector, gotta install em all 😄  ) and rebuild virtualenv with it.

> Should we pin the Python version and numpy version in that case?

If u pin to Python3.8 and numpy < 2.0 u need to remember to unpin as soon as the MuJoCo guys bump to numpy > 2.0 .

I suggest linking the information from this issues in README in order to point people to the solution for now.