MyoHub/myosuite #194: Cannot reproduce SAR-RL results in the MyoHand Reorient 100 task

Baliencasia, 2024-07-17:
A question that is almost identical to issue 133 [https://github.com/MyoHub/myosuite/issues/133](url)
[image]
This is the performance I am getting when using the SAR tutorial, ﻿ the red line represents the SAR trained from scratch, and the orange line represents using the precomputed SAR. 
Don't care about the starting point of the SAR, it's just a starting point that I manually set when drawing, representing the pretraining samples, the actual pretraining sample is consistent with the code, 1e6.
[image]
As you can see, the success metric of using SAR method is different from that in the paper, and the SAC seems to be better than the paper showed.
What caused this problem and how can I achieve better results?
﻿I found that in the video after the training, objects were easily dropped and the fingers did not seem to perform obvious reorientation operations. The successful attempts seemed to be when the target position was similar to the initial position, as shown in the video below.

https://github.com/user-attachments/assets/18357c59-c32c-47c8-9687-8099748416fc

Should I try reinstalling myosuite by cloning from GitHub, as mentioned in issue 133?

Thank you in advance for your help, and also thank you for developing the MyoSuite platform, which has promoted research in the fields of reinforcement learning, biomechanics, and musculoskeletal systems.

Baliencasia, 2024-08-06:
hello?

Vittorio-Caggiano, 2024-08-06:
hi @Baliencasia is this a regression with new versions of myosuite i.e. if you use older versions of the release e.g. 2.1.5, do you get different results?

Baliencasia, 2024-08-07:
hi @Vittorio-Caggiano the version i used is myosuite 2.1.5, stable-baselines3 1.6.1, gym 0.13.0. is that gymnasium should be used?  i also tried MyoSuite 2.4.0 (git clone) gymnasium 0.29.1 stable_baselines3 2.3.2, but the result didn’t change much. which version of gymnasium and sb3 may be useful?

closed 2024-09-07 by Baliencasia