MyoHub/myosuite #200: Loading policy looks for mujoco-py instead of mujoco

irfanmohamed12, 2024-08-02:
I cant seem to run the tutorial 2_Load_Policy.ipnyb. 

The issue throws an error at this line:
pi = pickle.load(open(policy, 'rb')).

The error 'DependencyNotInstalled: No module named 'mujoco_py'.

So I did a pip install mujoco_py.

If I run the notebook again, I see error at same line: ' You appear to be missing MuJoCo.  We expected to find the file here: 
C:\Users\USERNAME\.mujoco\mujoco210'

Does this mean I need to download Mujoco from https://www.roboti.us/download.html and save to this folder?

However, https://github.com/openai/mujoco-py#install-mujoco states that this pipeline is for Linux or OSX.

Is this a known issue? How do i solve it for Windows?

**Related info:**
Windows 11 Enterprise
python 3.9.13
Myosuite 2.50.
mujoco 3.1.2
mrjl 1.0.0
mujoco-py 2.1.2.14

Vittorio-Caggiano, 2024-08-02:
how did you install `mjrl`? 
can you try creating a new conda environment without `mujoco_py` and installing mjrl with `git+https://github.com/aravindr93/mjrl.git@pvr_beta_1vk` (in this branch the dependency on `mujoco_py` was removed)

Vittorio-Caggiano, 2024-08-02:
@irfanmohamed12 see issue #88

Vittorio-Caggiano, 2024-08-06:
@irfanmohamed12 did that solution fix your problem?

TaoshuaiZ, 2025-05-07:
> how did you install `mjrl`? can you try creating a new conda environment without `mujoco_py` and installing mjrl with `git+https://github.com/aravindr93/mjrl.git@pvr_beta_1vk` (in this branch the dependency on `mujoco_py` was removed)

Thank you. It is helpful!