MyoHub/myosuite #201: 'NoneType' object has no attribute 'glGetError'

XiaobenLi00, 2024-08-05:
I try to run the demo in the tutorials, but I find the following problem
[image]
I have tried the solutions in https://github.com/facebookresearch/AnimatedDrawings/issues/99 but it sames not work.
Looking forward to your help!

Vittorio-Caggiano, 2024-08-05:
can you share what OS and python version?

XiaobenLi00, 2024-08-06:
I am using Docker, and my OS is 22.04 LTS
[image]
My python version is 3.8

XiaobenLi00, 2024-08-06:
I retried but the ipykernel died.
[image]

Vittorio-Caggiano, 2024-08-06:
we have used docker for the myochallenge (see [here](https://github.com/MyoHub/myochallenge_2023eval/blob/main/docker/agent/Dockerfile_Mani_Random)) and worked with a very simple setting

can you please share your docker container setup to replicate the issue?

XiaobenLi00, 2024-08-06:
I used the docker with cuda11.8_conda3_unbuntu22.0, and create container from it. Actually I can run the test `python -m myosuite.tests.test_myo` but have problem with headless rendering with opengl to visualize.

XiaobenLi00, 2024-08-06:
And I installed myosuite following the advanced instruction.

Vittorio-Caggiano, 2024-08-06:
for our [CI test](https://github.com/MyoHub/myosuite/blob/ec185ab58afc26711a2f18fd3e6b0cf817964f42/.github/workflows/python-app.yml#L58) we use
``` bash
sudo apt-get install -y xvfb x11-xserver-utils
sudo /usr/bin/Xvfb $DISPLAY -screen 0 1280x1024x24 &
```
did you try that option?

XiaobenLi00, 2024-08-06:
[image]
I just tried and I got another problem.

Vittorio-Caggiano, 2024-08-22:
@XiaobenLi00 did you find a solution? or do you have more info on the issue to help troubleshooting?

closed 2024-08-23 by XiaobenLi00