MyoHub/myosuite #202: Questions about the manipulation task in myochallenge 2024

Soheavyrain, 2024-08-07:
I was wondering if it would be possible to modify the reward function for the task. I believe that with a tailored reward function, it might be possible to achieve better performance for specific goals.
Is it allowed to adjust the reward function in the myochallenge 2024?

Soheavyrain, 2024-08-07:
And I noticed that in the “myoChallengeBimanual-v0”, the DEFAULT_OBS_KEYS are not fully aligned with the “obs_dict”. I was wondering if it is permissible to adjust the DEFAULT_OBS_KEYS to better suit my requirements.

Would making such a modification be within the competition rules, or are there specific constraints I should be aware of?

raku-slyu, 2024-08-07:
Hello:

The reward function is there to provide a guidance. Please feel free to propose changes if that fits into the task better.

Regarding the observations, you are encouraged to study the behaviour of the environment in any possible ways, but in the final submission stage, you will only have access to the default observations.

I hope this clarifies the question

jamesheald, 2024-08-07:
> And I noticed that in the “myoChallengeBimanual-v0”, the DEFAULT_OBS_KEYS are not fully aligned with the “obs_dict”. I was wondering if it is permissible to adjust the DEFAULT_OBS_KEYS to better suit my requirements.

It's a shame the non-default observations can't be utilized (optionally). It might allow a wider array of strategies to be explored.

Soheavyrain, 2024-08-08:
ok, thank you.

cherylwang20, 2024-08-08:
Hi all, 

Thank you for your input. The default observations are usually not altered because they allow for standard evaluation of each policy without having to change it to each participant explicitly on the EvalAI platform. This has been the case for the past two challenges.

That being the case, the current default observations already include sufficient information on myoArm, MPL, objects and even touching. We'd like to encourage you to make full use of those. To clarify, you can freely utilize the `obs_dict` with all the keys and compute new entries from existing keys to design your reward. However, the simulator will not have access to those. 

Hope this clarifies things on top of @raku-slyu 's response.

closed 2024-08-08 by cherylwang20

jamesheald, 2024-08-09:
Hey @cherylwang20. Thanks for the clarification. Having access to the non-default observations during training is very useful, I agree.