MyoHub/myosuite #207: AttributeError: 'TimeLimit' object has no attribute 'get_obs', How to solve this problem？

llll111111, 2024-08-12:


Vittorio-Caggiano, 2024-08-12:
hi @llll111111 can you please add information on OS, python version and myosuite version? Also, a snippet of code to reproduce the problem?

Vittorio-Caggiano, 2024-09-10:
thanks @gandie ... may I ask you to provide a minimal example? it seems you might have already a fix too... happy to review a PR too :)

gandie, 2024-09-10:
> thanks @gandie ... may I ask you to provide a minimal example? it seems you might have already a fix too... happy to review a PR too :)

Hm i think it's rather a mistake in usage than something wrong with the code here. I can produce similar looking errors by building a `SubprocVecEnv` in a bad way:

```python
env = SubprocVecEnv([gym.make('myoLegWalk-v0') for i in range(24)])
```

... will run into:

```
Traceback (most recent call last):
  File "C:\Program Files\Python38\lib\multiprocessing\process.py", line 315, in _bootstrap
    self.run()
  File "C:\Program Files\Python38\lib\multiprocessing\process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\lars\myosuite\venv\lib\site-packages\stable_baselines3\common\vec_env\subproc_vec_env.py", line 29, in _worker
    env = _patch_env(env_fn_wrapper.var())
TypeError: 'TimeLimit' object is not callable
```

When we create an arbitrary env in Python shell:

```python
from myosuite.utils import gym
env = gym.make('myoLegWalk-v0')
```

... and check the wrapped environment we get:

```python
>>> env
<TimeLimit<OrderEnforcing<PassiveEnvChecker<WalkEnvV0<myoLegWalk-v0>>>>>
```

Btw `TimeLimit` wrapper is automatically added due to `max_episode_steps` setting set.
We can then call ...

```python
>>> env.get_obs
C:\Users\lars\myosuite\venv\lib\site-packages\gymnasium\core.py:311: UserWarning: WARN: env.get_obs to get variables from other wrappers is deprecated and will be removed in v1.0, to get this variable you can do `env.unwrapped.get_obs` for environment variables or `env.get_wrapper_attr('get_obs')` that will search the reminding wrappers.     
  logger.warn(
<bound method MujocoEnv.get_obs of <myosuite.envs.myo.myobase.walk_v0.WalkEnvV0 object at 0x000001F03CB14CA0>
```

and even get a warning to avoid grabbing into env this way.

Sooo, to round things up i'd say if you encountered `AttributeError` on `TimeLimit` ( which is the uppermost wrapper in vanilla myo environments if i see this correctly ) you most probably reach into environment without respecting wrappers. That what i've learned from my own happy accidents 😄