MyoHub/myosuite #209: Suggestions on integrating exokseleton to the legged model

NuCapybara, 2024-08-15:
Hi MyoSuite Contributers,

Thank you so much for answering my questions first!
I am a research intern trying to integrate the MyoSuite leg model into a lower-body exoskeleton model. I have gone through the models and found the models in "/myosuite/myosuite/simhive/myo_sim/leg/myolegs.xml". However, this model is wrapped with some geoms and also coming with the background rather than purely muscuskeleton itself. May I ask if there is anyway that I could find a pure muscuskeleton model with leg muscles so that I could integrate into the exoskeleton?
Besides of that, may I ask if you could have any other suggestions before integration?

Thanks again for answering my questions!!

[image]

Vittorio-Caggiano, 2024-08-16:
Hi @NuCapybara , thanks for your interest in MyoSuite! 

I think you are almost there: if you go in the `/myo_sim/leg/assets` folder you can find the assets file which give you access to the information how the body is build. From there you should be able to add your exoskeleton. 

If you need help with models specifically you can open an issue and/or make a draft PR in the [myo_sim repo](https://github.com/MyoHub/myo_sim) and we can help you. Alternatively, if you just want some general feedback, you can use the discussion groups: https://github.com/MyoHub/myosuite/discussions

NuCapybara, 2024-08-16:
> Hi @NuCapybara , thanks for your interest in MyoSuite!
>
> I think you are almost there: if you go in the `/myo_sim/leg/assets` folder you can find the assets file which give you access to the [...]

Thank you so much!

closed 2024-08-17 by Vittorio-Caggiano