MyoHub/myosuite #210: reset() function and render_mode problem

XiaobenLi00, 2024-08-18:
when I run [myosuite/agents/sb3_job_script.py](https://github.com/MyoHub/myosuite/blob/main/myosuite/agents/sb3_job_script.py)
I got this
```
WARN: The environment is being initialised with render_mode='human' that is not in the possible render_modes ([]).
Warning: Unused kwargs found: {'render_mode': 'human'}
WARN: The obs returned by the `reset()` method was expecting numpy array dtype to be float32, actual type: float64
WARN: The obs returned by the `reset()` method is not within the observation space.
```
I am wondering why got these warnings and how to solve them

XiaobenLi00, 2024-08-18:
For the reset() problem, it can be solved by a recent PR https://github.com/MyoHub/myosuite/pull/205,

but for the render_mode problem is still not solved, will you fix this problem?

jamesheald, 2024-08-22:
Hi @XiaobenLi00.

As mentioned [here](https://github.com/MyoHub/myosuite/issues/212#issuecomment-2305516981), in myosuite, onscreen and offscreen rendering is performed by calling env.mj_render() and env.sim.renderer.render_offscreen(), respectively. The render_mode is therefore not needed and env.metadata['render_modes'] is consequently an empty list. This is different to many gym environments where rendering is performed by calling env.render(), and the desired type of rendering is specified by the render_mode (e.g. 'human' or 'rgb_array'), which is passed to gym.make().

XiaobenLi00, 2024-08-23:
Thanks a lot for your answer!

closed 2024-08-23 by XiaobenLi00