MyoHub/myosuite #211: How to load a JSON file of a customized 3D model to the Myosuite Environment?

Escoastin, 2024-08-19:
I possess an OpenSim model I successfully converted from  .OSIM to .XML then to .JSON (It's a neuromuscular model). How do I upload it to Myosuite and save it as an environment so that I can play with it?

Vittorio-Caggiano, 2024-08-19:
Hi @Escoastin how did you convert the model from Opensim to xml? Did you use the [MyoConverter](https://github.com/MyoHub/myoconverter)?

Vittorio-Caggiano, 2024-08-19:
Please, have a look at the [repository of musculoskeletal models ](https://github.com/MyoHub/myo_sim) we use in MyoSuite

Escoastin, 2024-08-19:
> Hi @Escoastin how did you convert the model from Opensim to xml? Did you use the [MyoConverter](https://github.com/MyoHub/myoconverter)?

I used the convert files feature in OpenSim and it transformed it into XML. I then proceeded to use a XML to json converter with Miniconda
[image]

closed 2024-08-21 by Vittorio-Caggiano

Escoastin, 2024-08-21:
> I don't think Opensim supports natively the conversion of their models into MyoSuite format ... you might need to use the [MyoConverter](https://github.com/MyoHub/myoconverter)

Hello again. 

I've tried using the myoconverter however it seems as if there are some missing geometry attributes to my model. I've opened up an issue in the myoconverter forum but have had no response so far. Does the conversion work with personal models, or just the ones in the Myoconverted library?

Vittorio-Caggiano, 2024-08-21:
MyoConverter should work on any opensim model 4.0 that complies with the conversion format (see [limitations](https://myoconverter.readthedocs.io/en/latest/limitations.html))