MyoHub/myosuite #221: Record videos during training

XiaobenLi00, 2024-09-04:
As discussed in https://github.com/MyoHub/myosuite/issues/212, I tried to modify `VideoRecorderCallback` from https://stable-baselines3.readthedocs.io/en/master/guide/tensorboard.html  using `env.sim.renderer.render_offscreen() ` as follows:
```python
class VideoRecorderCallback(BaseCallback):
    def __init__(self, eval_env: gym.Env, render_freq: int, n_eval_episodes: int = 5, deterministic: bool = True):
        """
        Records a video of an agent's trajectory traversing ``eval_env`` and logs it to TensorBoard

        :param eval_env: A gym environment from which the trajectory is recorded
        :param render_freq: Render the agent's trajectory every eval_freq call of the callback.
        :param n_eval_episodes: Number of episodes to render
        :param deterministic: Whether to use deterministic or stochastic policy
        """
        super().__init__()
        self._eval_env = eval_env
        self._render_freq = render_freq
        self._n_eval_episodes = n_eval_episodes
        self._deterministic = deterministic

    def _on_step(self) -> bool:
        if self.n_calls % self._render_freq == 0:
            screens = []

            def grab_screens(_locals: Dict[str, Any], _globals: Dict[str, Any]) -> None:
                """
                Renders the environment in its current state, recording the screen in the captured `screens` list

                :param _locals: A dictionary containing all local variables of the callback's scope
                :param _globals: A dictionary containing all global variables of the callback's scope
                """
                # We expect `render()` to return a uint8 array with values in [0, 255] or a float array
                # with values in [0, 1], as described in
                # https://pytorch.org/docs/stable/tensorboard.html#torch.utils.tensorboard.writer.SummaryWriter.add_video
                # screen = self._eval_env.render(mode="rgb_array")
                screen = self._eval_env.unwrapped.sim.renderer.render_offscreen(
                        width=640,
                        height=480,
                        camera_id=1)
                # PyTorch uses CxHxW vs HxWxC gym (and tensorflow) image convention
                screens.append(screen.transpose(2, 0, 1))

            evaluate_policy(
                self.model,
                self._eval_env,
                callback=grab_screens,
                n_eval_episodes=self._n_eval_episodes,
                deterministic=self._deterministic,
            )
            self.logger.record(
                "trajectory/video",
                Video(torch.from_numpy(np.asarray([screens])), fps=40),
                exclude=("stdout", "log", "json", "csv"),
            )
        return True

```

But it seems stuck in rendering, do you have any suggestions? Or could you considering provide a usable one?

Page-David, 2025-04-23:
Hi, your previous discussion helped me a lot. Here is how I record videos:

1. Add VideoRecorderCallback callback example provided from SB3 at the end of `in_callbacks.py` without any modification.
2. At the beginning of `sb3_job_script.py`, add the `RenderWrapper` class mentioned in [last discussion](https://github.com/MyoHub/myosuite/issues/212#issuecomment-2629448948)
3. In `train_loop` function in `sb3_job_script.py`, modify the env definition:
```Python3
    # Create the vectorized environment and normalize ob
    env_id = job_data.env
    env = make_vec_env(env_id, n_envs=job_data.n_env)
    env = VecNormalize(env, norm_obs=True, norm_reward=False, clip_obs=10.)

    eval_env = make_vec_env(lambda: RenderWrapper(gym.make(env_id)), n_envs=job_data.n_eval_env)
    eval_env = VecNormalize(eval_env, norm_obs=True, norm_reward=False, clip_obs=10.)
```
4. Finally, add callback `callback += [VideoRecorderCallback(eval_env, render_freq=max(job_data.render_freq // job_data.n_env, 1))]`

After doing these, gif during evaluation will be produced in wandb.

Vittorio-Caggiano, 2025-04-23:
@Page-David this is great! do you want to start a PR to fix this in the `sb3_job_script.py` ?