MyoHub/myosuite #225: Interact with MyoSkeleton

decomiteA, 2024-09-08:
I saw that the last release to date (Myosuite 2.7.0) included the Myoskeleton model that I am pretty excited to with. However I could not find it in the available environments and missed the instructions about how to use it. 
Am i missing something ? 

Thanks,

Antoine

Vittorio-Caggiano, 2024-09-12:
@decomiteA this is just an initial integration to visualize and interact with the MyoSkeleton. Actuation and tasks will come soon

fl0fischer, 2026-10-05:
The MyoSkeleton is not a gym env; tutorial 3.5 (`tutorials/3.5_Playback_Mot_File.ipynb`) shows how to load it and play back an OpenSim `.mot` motion on it. Closing as answered; please reopen if something is missing.

closed 2026-10-05 by fl0fischer