MyoHub/myosuite #238: Locomotion gait trajectory

scakki, 2024-09-22:
Hello,

First of all, this repository is very helpful, and I appreciate the resources provided. I'm currently looking for gait trajectory data similar to [sample_gait_cycle.csv](https://github.com/MyoHub/myosuite/blob/60a1d42ea7e0b95dcfa75977488ff2a21a3cb8aa/myosuite/envs/myo/assets/leg/sample_gait_cycle.csv). Could you suggest where I might find more compatible data for use with the myoleg or myoconverter lower extremities model?

Thank you!

elladyr, 2024-09-26:
Hi @scakki, just a disclaimer about the sample data. It is just some sample data I captured of myself and is meant as a demo, not as a representation of a person walking with a prosthetic leg.

Some dataset you could consider could be the Mixamo dataset by Adobe. This dataset was used by locomotion winner of MyoChallenge 2023 for their imitation learning algorithm.

If you are searching for datasets captured from human studies, we are currently in process of putting together a database of links, but this is still a work-in-progress and will not be available in the immediate future.

scakki, 2024-09-26:
Hello @elladyr, thank you for your response! Yes, I am indeed looking for datasets captured from human studies—specifically, human walking joint position data similar to the sample demo (not involving a prosthetic leg). I also appreciate the recommendation of the Mixamo dataset, which I am currently exploring. I’m particularly interested in how it can be adapted for use with the MyoLegs model.

If you have any additional suggestions or resources that could be helpful, I would greatly appreciate it!

Vittorio-Caggiano, 2024-09-26:
@scakki depending on the size of the dataset you are looking for, you could look also into opensim data formatted for the same leg model e.g. [Rajagopal et al. (2016) or 2392 tutorial](https://github.com/opensim-org/opensim-models/blob/c62c24b0da1f89178335cf10f646a39c90d15580/notebooks/resources/Tutorial%204/normal_gait.mot)