MyoHub/myosuite #244: Issue with rendering when running the installation example

ViktorM, 2024-09-24:
Running the default example the installation got this error on RTX 4090 laptop and Ubuntu 24.04. Any recommendations of how to fix it?

```
python -m myosuite.utils.examine_env --env_name myoElbowPose1D6MRandom-v0
MyoSuite:> Registering Myo Envs
    MyoSuite: A contact-rich simulation suite for musculoskeletal motor control
        Vittorio Caggiano, Huawei Wang, Guillaume Durandau, Massimo Sartori, Vikash Kumar
        L4DC-2019 | https://sites.google.com/view/myosuite
    
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
/home/viktor4090/anaconda3/envs/myosuite/lib/python3.10/site-packages/glfw/__init__.py:914: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
ERROR: could not create window
```

Vittorio-Caggiano, 2024-09-24:
Hi @ViktorM , this seems a problem of mujoco rendering. In this issue on the mujoco repo https://github.com/google-deepmind/mujoco/issues/692  they suggest this [stackoverflow solution](https://stackoverflow.com/questions/72110384/libgl-error-mesa-loader-failed-to-open-iris). Can you check if that works for you too?

ViktorM, 2024-09-24:
Thank you! I just found this thread on stack: https://stackoverflow.com/questions/71010343/cannot-load-swrast-and-iris-drivers-in-fedora-35/72200748#72200748 and this advice from there helped me:

```
I added the export LD_PRELOAD to my .bashrc file and it's working fine.

export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6
```

Vittorio-Caggiano, 2024-09-24:
Great! Thanks @ViktorM !

closed 2024-09-24 by Vittorio-Caggiano