MyoHub/myosuite #245: Shape of obs_dict["pros_hand_qpos"]

XiaobenLi00, 2024-09-25:
When I use trained model to eval on the updated env, I got following error:

`ValueError: Error: Unexpected observation shape (210,) for Box environment, please use (211,) or (n_env, 211) for the observation shape.`

I notice that this might be caused by the shape of `obs_dict["pros_hand_qpos"] `, which is 27 * 1 in https://github.com/MyoHub/myosuite/blob/3281da5c665fcb3a114ad7f9822c67399f677acf/myosuite/envs/myo/myochallenge/bimanual_v0.py#L158 and 26 * 1 in https://github.com/MyoHub/myosuite/blob/96fae9cffea2152dd06c26ce775d088bb57cb2cc/myosuite/envs/myo/myochallenge/bimanual_v0.py#L194
I guess the latter one is right, is it? Then how should I fix this error to make my trained agents compatible with the updated env?

cherylwang20, 2024-09-25:
Hi Xiaoben, 

Thank you for your question. We indeed changed the observation space and we understand the inconvenience that this may cause. This change is due to the fact that in the previous observation state, the object qpos is counted twice (one is object qpos, and one is accidentally attached at the end of MPL qpos).

Hence, we would like to provide you with the script to convert your previous policy to be compatible with the new observation state. Are you using SB3? This script is used for sb3 conversion but can serve as a reference for all other RL implementations. This is what you need to do:

1. Decompress the zip file of your policy, we call it here `edit_baseline`
2. Create a folder name `new_baseline` for your converted policy. 
2. Follow the script `'weight change.ipynb'` we provided to change the weights of the duplicate observations and delete one of them. This should save two new files: `policy.optimizer.pth` and `policy.pth` in  `new_baseline` folder
3. copy the remaining four files `_stable_baselines3_version`, `data`, `pytorch_variables.pth`, `system_info` from `edit_baseline` folder to `new_baseline`.
4. Replace the `observation_space' dict in data file with:  
```
    "observation_space": {
        ":type:": "<class 'gymnasium.spaces.box.Box'>",
        ":serialized:": "gAWVrwkAAAAAAACMFGd5bW5hc2l1bS5zcGFjZXMuYm94lIwDQm94lJOUKYGUfZQojAVkdHlwZZSMBW51bXB5lGgFk5SMAmY0lImIh5RSlChLA4wBPJROTk5K/////0r/////SwB0lGKMDWJvdW5kZWRfYmVsb3eUjBJudW1weS5jb3JlLm51bWVyaWOUjAtfZnJvbWJ1ZmZlcpSTlCiW0gAAAAAAAAABAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQGUaAeMAmIxlImIh5RSlChLA4wBfJROTk5K/////0r/////SwB0lGJL0oWUjAFDlHSUUpSMDWJvdW5kZWRfYWJvdmWUaBAoltIAAAAAAAAAAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBAQEBlGgUS9KFlGgYdJRSlIwGX3NoYXBllEvShZSMA2xvd5RoECiWSAMAAAAAAAAAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMEAACDBAAAgwQAAIMGUaApL0oWUaBh0lFKUjARoaWdolGgQKJZIAwAAAAAAAAAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQQAAIEEAACBBAAAgQZRoCkvShZRoGHSUUpSMCGxvd19yZXBylIwFLTEwLjCUjAloaWdoX3JlcHKUjAQxMC4wlIwKX25wX3JhbmRvbZROdWIu",
        "dtype": "float32",
        "bounded_below": "[ True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True\n  True  True  True  True  True  True  True  True  True  True  True  True [... cut ...]
        "low_repr": "-10.0",
        "high_repr": "10.0",
        "_np_random": null
    }
```
5. create a new zip with the six files in your `new_baseline` folder. 
6. Load the policy into sb3 just like your previous policy. 

Please let us know if you encounter any issues along the way and we are happy to help. We used this method to convert our baseline, and the performance stays the same. 

Best, 
Cheryl
[weight_change.zip](https://github.com/user-attachments/files/17139389/weight_change.zip)

Balint-H, 2024-09-26:
An alternative solution that you can consider is simply applying a preprocessing wrapper to your policy, that repeats the missing observation, extending the vector to its original size. Namely you would need to append copy the first element of `obs_dict["object_qpos"] `, and append it to the end of `obs_dict["pros_hand_qpos"]`. This may be a bit faster to implement, but is a bit of a hack though, and the repeated value introduces a small amount of wasted operations in learning based policies, which is why we applied the fix in the first place. Let us know if either option solves your issue!

cherylwang20, 2024-10-07:
Closing this for now. Please reach out to us if you have any further questions.

closed 2024-10-07 by cherylwang20