MyoHub/myosuite #246: unknown file '/home/yyb/myosuite-main/myosuite/envs/myo/myobase/../assets/elbow/../../../../simhive/myo_sim/elbow/assets/myoelbow_assets.xml

c1299877959, 2024-09-26:
how can I solve this problem，please

raku-slyu, 2024-09-26:
Hello:

Can you check the relevant path of "myosuite/simhive/myo_sim" folder and whether it is empty. It might be the submodule that is not properly cloned. It that's the case, you can use the "recursive" clone to fix this issue.

Hope that helps

c1299877959, 2024-09-26:
Thank you very much for your guidance. The error has been resolved. The Myo Sim folder is the only non-empty folder in the SimHive directory. Do I also need to recursively download the other empty folders in the directory?

c1299877959, 2024-09-26:
Thank you very much for your guidance; the error has been resolved. The other folders under the simhive directory are also empty. Do I need to recursively download them as well?

raku-slyu, 2024-09-26:
> Thank you very much for your guidance; the error has been resolved. The other folders under the simhive directory are also empty. Do I need to recursively download them as well?

So the submodules contains assets files for running myosuite environments and it is recommended to clone all of them to obtain the best experience

c1299877959, 2024-09-26:
Sorry to bother you again, when I run this line of command:python -m myosuite.utils.examine_env --env_name myoElbowPose1D6MRandom-v0  , the pop-up window will blink back in about 20s, how should I solve the problem? 

The following errors are reported：
Average success over rollouts: 70.0%
Exception ignored in: <function Renderer.__del__ at 0x7fef1197b550>

raku-slyu, 2024-09-26:
You can find the source code of the command [here](https://github.com/MyoHub/myosuite/blob/main/myosuite/utils/examine_env.py). The env is scheduled to terminate in around 20 seconds. The exception seems to be an issue with the renderer, but it doesn't seem to affect the program performance.

closed 2025-04-11 by Vittorio-Caggiano