MyoHub/myosuite #263: Inverse dynamics tutorial

alco24, 2024-10-30:
Hi MyoSuite team, 
I'm learning how to use MyoSuite for biomechanical studies and musculoskeletal simulations and I was trying to adapt the Inverse dynamics tutorial to the MyoLeg model. I have problems in computing the generalized force and I think it's because the number of generalized coordinates (dim(qpos)) does not match the number of degrees of freedom (dim(qvel)), because of the presence of a free joint ('root') in the myoleg model. Am I missing something? How can I overcome this problem? Thank you very much for your job here!

Beanpow, 2024-11-04:
Please refer to https://github.com/google-deepmind/mujoco/issues/645

andreh1111, 2024-11-04:
Hi @alco24, you are right to note that the problem is related to the presence of the free joint root, the handling of which is indeed not supported by the current version of inverse dynamics in the tutorial. However, the main problem is not the different dimensionality resulting from the use of quaternions in qpos, which can indeed be addressed in multiple ways, but rather the zero actuator_moment for the free joint, since the muscle actuators do not act directly on it.

closed 2026-10-07 by fl0fischer