MyoHub/myosuite #268: Inverse kinematics

alco24, 2024-11-11:
Hi MyoSuite team, 
can you give me an input to perform inverse kinematics with MyoSuite? I have my 3D coordinates, but is there an algorithm to perform inverse kinematics in MyoSuite? Thank you!

jamesheald, 2024-11-13:
Hey @alco24, does [this](https://github.com/kevinzakka/mink?tab=readme-ov-file) do what you need? @Vittorio-Caggiano  [recommended](https://myosuite.slack.com/archives/C044SDE0XV2/p1730120400231469) it on the MyoSuite slack workspace, saying "the results are impressive: faster-than-realtime differential inverse kinematics (dIK) achieved!".

Vittorio-Caggiano, 2025-04-11:
Please, see this example https://github.com/MyoHub/myosuite/blob/dev/docs/source/tutorials/8_inverse_kinematics.py

closed 2025-04-11 by Vittorio-Caggiano