MyoHub/myosuite #272: Full-body Musculoskeletal Model

acrlw, 2024-12-24:
Dear MyoSuite Team,

I would like to know if there is a plan to release a full-body musculoskeletal model. I noticed a similar request in issue #173  , but when I checked version 2.7.0 of MyoSuite, I found only a full-body skeletal model without muscle control.

Thank you!

valaxkong, 2025-02-07:
I have the same doubt. I found that the MyoLeg model is the closest to the Full-body muscle-skeleton model, but it seems to only involve the lower-body muscles.

ajaytalati, 2025-02-09:
I would guess that adding a full body muscular system (like OpenSim) is huge task? 

Maybe though just releasing the muscles which articulate the whole spine would be an easier project?

This would very helpful for postural research. It would also make a lot of more interesting tasks possible (i.e. muscular control to prevent spinal pathologies like scoliosis and kyphosis)?

Vittorio-Caggiano, 2025-02-09:
we have released [myolegs with abdomen](https://github.com/MyoHub/myo_sim/tree/main/leg) and we in the process to release a fully articulated lumbar back model. Hope those help

ajaytalati, 2025-02-26:
Hi @Vittorio-Caggiano 

i was just wondering if the MyoBack model is likely to be released soon? I saw in the pull request that @cherylwang20 mention,

I also have created an env to achieve a simple 'upright' position for this model. Do you want me to add it as well?

that would be great if it was released as well

All the best,

Ajay

cherylwang20, 2025-03-01:
> Hi [@Vittorio-Caggiano](https://github.com/Vittorio-Caggiano)
>
> i was just wondering if the MyoBack model is likely to be released soon? I saw in the pull request that [...]

Hi @ajaytalati Thanks for your question. We are in the processing of releasing the model and the base env. Please expect an update in the next week or two.

sharinka0715, 2025-08-04:
Hi MyoSuite Developers @Vittorio-Caggiano @cherylwang20,

I am writing to express my strong interest in a complete, full-body musculoskeletal model. Implementing such a model in the MyoSuite simulator would undoubtedly enable the development of many impressive control strategies.

Unfortunately, it seems there is currently no **open-source, complete, muscle-actuated** model that can **run in MuJoCo**.
* [MS-HUMAN-700](https://lnsgroup.cc/research/MS-Human-700) is not yet open-source.
* [MyoSkeleton](https://github.com/myolab/myo_model) lacks muscle actuators, and while other models in MyoSuite cover various body parts, they do not constitute a single, complete model.
* [MASS](https://github.com/lsw9021/MASS) is based on the DART simulator, which has several limitations in terms of performance and generalizability. Similarly, the numerous biological models in [OpenSim](https://simtk.org/projects/opensim/) cannot be directly used in MuJoCo.

I believe MyoSuite is an excellent community with a well-developed ecosystem and a reliable advocacy group, and I am confident that work capable of advancing the field of musculoskeletal control will emerge from a platform based on MyoSuite. Although efforts such as [MuscleVAE](https://pku-mocca.github.io/MuscleVAE-page/) and [MAGNET](https://dl.acm.org/doi/10.1145/3721238.3730617) have been made to develop impressive whole-body control policies, their impact remains limited. Leveraging [MuJoCo's MJX](https://mujoco.readthedocs.io/en/latest/mjx.html) or even the [MuJoCo Warp](https://github.com/google-deepmind/mujoco_warp) framework could significantly accelerate research in this area. For instance, projects like [Kinesis](https://github.com/amathislab/Kinesis) and [Loco-MuJoCo](https://github.com/robfiras/loco-mujoco) have already demonstrated remarkable performance using MyoSuite's half-body models.

I would like to inquire if the MyoSuite maintenance team has any plans to create and open-source a full-body musculoskeletal model.

I sincerely look forward to your reply. Thank you!

Vittorio-Caggiano, 2025-08-05:
> Hi MyoSuite Developers [@Vittorio-Caggiano](https://github.com/Vittorio-Caggiano) [@cherylwang20](https://github.com/cherylwang20),
>
> I am writing to express my strong interest in a complete, [...]

Dear @sharinka0715 
Thank you for the detailed and insightful message. We really appreciate your interest in MyoSuite and your thoughtful overview of the current landscape.

@cherylwang20  has been actively building a full-body musculoskeletal model compatible with the myoskeleton, and it’s almost ready. We can point you to an early preview version soon (cc: @cherylwang20 ), and we would really value your feedback and contributions to help us iterate and finalize it.
In the meanwhile, did you start testing the myoskeleton? Do you have any feedback about it? Please, do not refrain to make a PR or raise any issue.

Thanks again for supporting the MyoSuite community!

sharinka0715, 2025-08-05:
> Dear [@sharinka0715](https://github.com/sharinka0715) Thank you for the detailed and insightful message. We really appreciate your interest in MyoSuite and your thoughtful overview of the current [...]

Hello @Vittorio-Caggiano,

I haven't tested MyoSkeleton myself, but I am aware that the [Loco-Mujoco](https://github.com/robfiras/loco-mujoco/) project has already implemented good support for it. This includes adding [torque-controlled actuators](https://github.com/robfiras/loco-mujoco/blob/131c1e7bfb4c3d20ffc82c0368886fe708a5964b/loco_mujoco/environments/humanoids/myoskeleton.py#L520) to each joint and integrating them into their comprehensive motion and imitation learning framework as an optional model. They even provide [a dataset of retargeted LAFAN1 motions](https://huggingface.co/datasets/robfiras/loco-mujoco-datasets/tree/main/Lafan1/mocap/MyoSkeleton) for MyoSkeleton. I have not personally tried this part of the code, as my primary focus is on the performance of models under muscle actuation.

I would be thrilled to participate in any testing related to your full-body musculoskeletal model and look forward to your future releases.

fl0fischer, 2026-10-05:
In the upcoming 3.0 release (branch `ms3`, PR #406) full-body musculoskeletal models are supported: the MuscleMimic full-body env (354 muscles, `myoMimicFullbody-v0`) with trained policies, tutorials 5.1 to 5.5 and a baseline on Hugging Face (`myohub/myosuite-3-baselines`). Closing as done.

closed 2026-10-05 by fl0fischer