MyoHub/myosuite #276: Model Verification

JudyYe, 2025-02-26:
Daer team,

Thank you for your great efforts! I am working on a project that involves model verification between opensim and mujoco. 

In your [doc](https://github.com/MyoHub/myo_sim/tree/main/hand), myohand finger is optimized from [this opensim model](https://simtk.org/projects/hand_muscle), converted from [myoconverter](https://github.com/MyoHub/myoconverter). However, when I run the verification step for myohand and the open sim model, the differences are quite significant. 

I also tried myoelbow 2dof 6muscle. They are more similar but it is not as similar as the model were directly converted by `myoconverter`. I wonder if there is anything that I missed when I run verification code? you also mentioned that you did some manual adjustment I wonder what are the adjustment for and if their consequences can explain the differences. Thank you! 

MyoElbow, plot from verification 3. more similar than hand model, but still not as similar as myoconverter's output. 
| MyoElbow vs OpenSim arm26.osim | myoconverter  vs arm26.osm | 
| --- | --- |
| [image] |[image]  |
| [image] | [image] | 

Myohand
| verification 2 | verification 3 | 
| --- | --- | 
| [image] |  [image]|

Vittorio-Caggiano, 2025-03-01:
> Daer team,
>
> Thank you for your great efforts! I am working on a project that involves model verification between opensim and mujoco.
>
> In your [doc](https://github.com/MyoHub/myo_sim/tree/main/hand), [...]

Hi @JudyYe , thanks for your interest in the myohand. 

Yes, we started from a few opensim models but after the conversion, we realized that 
1. the myohand model was quite unstable.
2. many muscles were not able to provide the right forces

This was due to different factors including different wrapping objects and different muscle models.  We started remodeling the muscles -- both attachments and wrapping surfaces -- to make the hand model stable and controllable.

fl0fischer, 2026-10-05:
This concerns the conversion and verification of the hand model, which live in `MyoHub/myo_sim` and `MyoHub/myoconverter`, not in the MyoSuite environments. Please raise it there. Closing here.

Vittorio-Caggiano, 2026-11-04:
Closing as answered. As explained in the maintainer response above, MyoHand was intentionally modified after its initial OpenSim conversion: muscle attachments and wrapping surfaces were remodeled to improve stability and force production. It should therefore not be expected to reproduce the source OpenSim model exactly. This closes the explanation request, not a claim of numerical or biomechanical equivalence; a specific remaining discrepancy can be tracked separately with the exact model versions and reproduction.

closed 2026-11-04 by Vittorio-Caggiano