MyoHub/myosuite #278: SAR_tutorial.ipynb experiences simulation instability

Bransthre, 2025-03-05:
Hello, I was running the `SAR_tutorial.ipynb` notebook for training with the following package versions as outlined in `requirements.txt`:
```
click==8.1.7
sk-video==1.1.10
gymnasium==0.29.1
termcolor==2.4.0
flatten_dict==0.4.2
h5py==3.12.1
numpy==1.26.4
mujoco==3.1.2
dm-control==0.1.16
pillow==10.4.0
pink-noise-rl==2.0.1
```
However, I frequently see the likes of following warning messages during training:
```
WARNING:absl:Nan, Inf or huge value in QVEL at DOF 0. The simulation is unstable. Time = 0.2180.
Simulation couldn't be stepped as intended. Issuing a reset
```
at, for example, section cells like "Option 2.1.A: Get SAR from scratch".

Is this an expected behavior? If not, should the requirements for SAR alone (published two years ago) be updated to some corresponding MuJoCo version?

Thanks in advance for the help!

Bransthre, 2025-03-11:
Hello, I have recently reverted to this commit ([link](https://github.com/MyoHub/myosuite/commit/c47c5366350ecc5f42fc3fd51afc7e25511210d6)) along with their package and environment settings. With stable-baselines, there is still a simulation instability problem that occurs in MyoHandReorient. Bizzarrely, that commit is precisely where the latest SAR tutorial update has been done. I believe this requires some investigation.

Thanks.

Vittorio-Caggiano, 2025-03-12:
Hi @Bransthre , you are completely right. This might be related to #267 too. Please, keep us posted in case you get more insights during your analysis