MyoHub/myosuite #295: No attribute 'playback' when using certain tasks

abhishekpatil32, 2025-03-26:
Hi @Vittorio-Caggiano and team,

First off, thank you for this amazing suite—it's been a fantastic tool for research and experimentation!

I'm currently working with tasks like myoHandKeyTurnFixed-v0 and myoHandPenTwirlFixed-v0, and I encounter the following error when trying to run a rendering loop:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
[<ipython-input-55-a1851fce0d2b>](https://localhost:8080/#) in <cell line: 66>()
     65 frames = []
     66 for i in range(env.spec.max_episode_steps):
---> 67     env.playback()
     68     frames.append(env.sim.renderer.render_offscreen(
     69                         width=400,

2 frames
[/usr/local/lib/python3.10/dist-packages/gymnasium/core.py](https://localhost:8080/#) in __getattr__(self, name)
    313             f"to get this variable you can do `env.unwrapped.{name}` for environment variables or `env.get_wrapper_attr('{name}')` that will search the reminding wrappers."
    314         )
--> 315         return getattr(self.env, name)
    316 
    317     def get_wrapper_attr(self, name: str) -> Any:

AttributeError: 'KeyTurnEnvV0' object has no attribute 'playback'
```

It seems like env.playback() is not available in these environments. Could you please advise on:

Whether playback() is deprecated or not implemented for these environments?

What the current recommended way is to render or visualize rollouts for these tasks?

Any help or guidance would be appreciated.

Thanks again for all your work!

Regards,
Abhishek

Vittorio-Caggiano, 2025-03-26:
Thanks @abhishekpatil32 !
Can you add information on which version of MyoSuite you are using and which operating system?

abhishekpatil32, 2025-03-26:
Thanks for the response!

Here are the details of my environment:

- MyoSuite version: 2.8.4
- Python version: 3.10.12
- Operating system: Google Colab (Linux)
- Gymnasium version: 0.29.1
- MuJoCo version: 3.1.2

Let me know if you need any additional logs or configuration details.

Vittorio-Caggiano, 2025-03-26:
I run this example on colab and it seems to render it correctly https://colab.research.google.com/drive/1OsbzTMOtBrrYU6cTJi7daw4nwwfq2Sw_?usp=sharing

abhishekpatil32, 2025-03-26:
Thanks for the quick response @Vittorio-Caggiano !!

I tried the example from the colab notebook that you shared, and it renders correctly now — everything seems to be working as expected.

Appreciate the support!

closed 2025-03-26 by Vittorio-Caggiano